commaai/openpilot #1707: Detect when Honda Bosch is in Cruise Mode

VirtuallyChris, 2020-06-15:
**Describe the bug**
A number of support emails have came through where people have their cars in `Cruise Mode` which is standard cruise control, yet openpilot says `Main Switch Off` which is incorrect.

**How to reproduce or log data**
Press and hold the Distance button to get the car into `Cruise Mode`
https://www.youtube.com/watch?time_continue=38&v=adV_qoCnJO0&feature=emb_title
Try to engage openpilot - it will say `Main Switch Off`

**Expected behavior**
Warning should say something like `ACC off, hold distance button to switch modes` - this would require detecting Cruise Mode and differentiating it from the Main Switch Off alert.

vanillagorillaa, 2020-06-17:
Didn't this user have a Nidec vehicle? I checked last weekend and at least on Nidec vehicles on start up we change to ACC mode. I was unable to get it start openpilot in Cruise mode. We should check on a Bosch vehicle

pd0wm, 2020-06-25:
Honda bosch was already locking you out, but with the wrong alert. Fixed by 611dc4c37f4a607c26542dce67d7568a8319cf55

closed 2020-06-25 by pd0wm