commaai/openpilot #1726: No clean way to shutdown openpilot on PC running with Carla

johschmitz, 2020-06-16:
**Describe the bug**
There seems to be no clean way to shut down openpilot when running on PC with Carla simulation.
It seems that one has to manually find and kill all the invoked processes.
Also bridge.py seems to be hijacking <kbd>Ctrl</kbd>+<kbd>C</kbd>. Instead it would be nice if it could be close with both <kbd>Q</kbd> and <kbd>Ctrl</kbd>+<kbd>C</kbd>.

**How to reproduce or log data**
Follow the steps in https://github.com/commaai/openpilot/blob/master/tools/sim/README.md
Then try to shutdown all processes.

**Expected behavior**
- Provide a _documented_ clean way of shutting down openpilot
- bridge.py should exit on <kbd>Ctrl</kbd>+<kbd>C</kbd>/SIGINT

**Device/Version information (please complete the following information):**
 - Device: PC
 - Version: 1d9920a1773041df9938e9f20066bbf35903f6bb (current master branch)

**Additional context**
Improving this will increase confidence in the quality of the software architecture.
Could be just missing documentation in https://github.com/commaai/openpilot/blob/master/tools/sim/README.md

ed2k, 2020-07-12:
make camerad a killable process, https://github.com/ed2k/openpilot/blob/test_keras/selfdrive/manager.py#L206

johschmitz, 2020-07-12:
Johannes Schmitz hat Ihnen eine E-Mail über den Gmail-Modus "Vertraulich" gesendet:

[image: Gmail-Logo]Re: [commaai/openpilot] No clean way to shutdown openpilot on PC running with Carla (#1726) <https://confidential-mail.google.com/msg/AA12eChaVZo7V0AXA4aLMyucA8ROcii8H7Nw2BHYh7mR2VRCfGCFn4soh9euued8iPuT-pAAoW93ZP7ICVLODkduhkt34rAnRN8zSz5QKEanxOat0xxJMGa8Ga8eYgvMl-hC-fyLSI4Hbg_AiR6yGYdhsSzptS-3EF4qZerXOkT7Rl7SNjWB5WZjNMSkbbsO2MxrvRgQ53Iz7J8brF0ebyyh>

Diese Nachricht wurde am 12.07.2020 um 11:07:41 GMT-7 gesendet
Zum Öffnen klicken Sie auf den unten angezeigten Link. Der Link kann nur von reply@reply.github.com geöffnet werden.

E-Mail ansehen <https://confidential-mail.google.com/msg/AA12eChaVZo7V0AXA4aLMyucA8ROcii8H7Nw2BHYh7mR2VRCfGCFn4soh9euued8iPuT-pAAoW93ZP7ICVLODkduhkt34rAnRN8zSz5QKEanxOat0xxJMGa8Ga8eYgvMl-hC-fyLSI4Hbg_AiR6yGYdhsSzptS-3EF4qZerXOkT7Rl7SNjWB5WZjNMSkbbsO2MxrvRgQ53Iz7J8brF0ebyyh>

Mit dem Gmail-Modus "Vertraulich" haben Sie mehr Kontrolle über Ihre gesendeten Nachrichten. Als Absender können Sie beispielsweise Ablaufdaten festlegen, das Drucken und Weiterleiten deaktivieren oder den Zugriff auf Nachrichten nachverfolgen. Weitere Informationen <https://support.google.com/mail/answer/7674059>
Gmail: E-Mail-Dienst von Google
Die Verwendung unterliegt der Google-Datenschutzerklärung <https://myaccount.google.com/privacypolicy?hl=de>
Google LLC, 1600 Amphitheatre Parkway, Mountain View, CA 94043, USA
Sie haben diese Nachricht erhalten, da Ihnen jemand eine E-Mail über den Gmail-Modus "Vertraulich" gesendet hat.
[image: Google-Logo]

PR #2035 (2020-08-16): exit camerad cleanly on PC
fixes #1726
closed 2020-08-16 by adeebshihadeh