commaai/openpilot #1732: Show alert if fan is not spinning

pd0wm, 2020-06-17:


gordio, 2020-06-19:
Fan without control wire, so maybe control overheat and show warning?

pd0wm, 2020-06-19:
The panda measures fan rpm. I was thinking an alert which shows if there is a command above 10% fan speed, but the rpm reads zero.

kishkaru, 2020-06-25:
> The panda measures fan rpm

How does it do this?

> an alert which shows if there is a command above 10% fan speed, but the rpm reads zero.

This will be annoying for those of us with Noctua fan mod, and soldered directly to 5V rails on EON board.
If possible, maybe a way to disable the alert? Or else the end user will need to edit the code to disable the alert.

LoganxDev, 2020-09-28:
I’m about to finish a first pass at this, what would be a good way to test this?

pd0wm, 2020-09-28:
You could make a change to boardd to ignore the fan command.

LoganxDev, 2020-09-28:
Sweet, I'll test it out tonight.

LoganxDev, 2020-10-03:
Opened #2258 for this.

closed 2020-10-07 by adeebshihadeh