commaai/openpilot #1733: Disengage immediately can error : check connections

adam-nelson, 2020-06-17:
I am finding since 0.7.6 has been released there been of increase.

> Disengage immediately can error : check connections

How can I help report and troubleshoot this to fix the error?

adam-nelson, 2020-06-17:
I am driving a toyota corolla hybrid 2019 hatch

adam-nelson, 2020-06-18:
18e8ac91b805db11|2020-06-18--07-48-13--29

pd0wm, 2020-06-19:
Looked at canErrors in the logs for all users, and there doesn't seem to be an increase related to 0.7.6. I seems more likely that your canErrors are hardware related, and you have a loose connection somewhere. I would recommend unplugging all your wires and plugging them back in.

An alternative cause could be cellular related lags when switching between certain cell towers. Does it always happen in the same location when driving?

adam-nelson, 2020-06-19:
Is there easy way to trace the cellular lag as possibility via cabana? A
wiki guide would be useful to write up

Driving more then 1-6 hour seem to trigger it.

pd0wm, 2020-06-20:
You can replay it using unlogger, and then run https://github.com/commaai/openpilot/blob/master/selfdrive/debug/filter_log_message.py to print the android log messages. The segment you posted seems to have some messages related to the modem just before the error happened.

We're working on a solution, but haven't figured it out yet. It seems like you're way more affected than average users. A temporary solution would be to remove your sim card.

adeebshihadeh, 2020-06-20:
Closing this, and reopening #1162

adam-nelson, 2020-06-22:
Thanks @pd0wm / William

jyoung8607, 2020-06-22:
[ reopening for the moment to work with this specific sub-case of #1162 ]

@adam-nelson, is this something you think is repeatable in the area that you drive through? We have something of a proxy/simulation here for your class of problem, which is showing us that 0.8 will perform much better. However, it would be nice to have A/B test proof it works for your exact situation, just in case our simulated test doesn't quite match your reality.

If you think this is repeatable, and you have the time and desire to run a couple of specific tests with me, I'd love to help make sure this is 100% resolved for you. If not, that's fine, you'll already be getting the probable fix in 0.8 when that's released.

adam-nelson, 2020-06-23:
I been doing passes on a daily basis in the same spot. Today it hasnt trigger an can error alarm

Problem is the scenario is different by the time i have time to test this after work hours and at night. Will that have an impact on these test if done at night? @jyoung8607

adam-nelson, 2020-06-23:
Closing for now. Will try master-ci for testing

closed 2020-06-23 by adam-nelson

NSExceptional, 2020-10-09:
This happens to me when I drive down Highway 6 going from Waco to Houston.

I noticed the device has no service out here. Is a cellular connection required for OpenPilot to work?!

adeebshihadeh, 2020-10-09:
@NSExceptional openpilot doesn't require a cell connection. what's your dongle ID and what version are you running? This should be reduced in 0.7.9 (released today), and a more complete fix will ship soon.

NSExceptional, 2020-10-10:
I do not know, I will get back to you on that.

I wonder if the update had anything to do with it? I briefly saw an update notification right before the ear-destroying "take control" sound distracted me the first time it happened on my trip today. I also noticed I got the CAN error whenever the device did not have service. I got it 3 separate times today in the span of about 15 minutes :/