commaai/openpilot #1740: Comma Two - Hyundai Kona Electric 2019 - "CAN communication has failure”

swells-Kona, 2020-06-18:
**Describe the bug**

Installed Openpilot on a new Comma Two that arrived yesterday.
Message onscreen is "CAN communication has failure"

**How to reproduce or log data**

Used a new install or openpilot.
Had followed instructions to get the latest from openpilot.comma.ai

**Expected behavior**
I think it should connect CAN communication

**Device/Version information (please complete the following information):**
 *Device: [Comma Two]
  *Version: [0.7.6-1]
  *Car make/model [Hyundai Kona Electric 2019 with LFA and HDA  ]

adeebshihadeh, 2020-06-19:
Can you post a route where this happens? You can see your routes with https://my.comma.ai.

crwusiz, 2020-06-25:
last week i tested kona ev 2019
use comma two
-- release2 can error
-> xx979xx HKG_community ver can error to 
-> add elect_ems and hyundaican add kona_ev no error work done..~

https://github.com/xx979xx/openpilot/pull/127

davidew98, 2020-06-26:
I have a Hyundai Kona Electric 2019 SEL (no SCC) and get similiar results:
- Original OP release2- Factory Reset, install openpilot.comma.ai, turn on community features, turn on car. Says: OP not available, buckle seat belt, Device Sensors not detected, CAN error. (fyi - Seat belt buckled) I even added elect_ems and added kona_ev to it but no help
-Installed HKG_community - looks like same fingerprint as original OP but won't detect and turn on OP like Original did. (this may be before the adds crwusiz did)
-In https://my.comma.ai/useradmin the release2 comes up as Hyundai Kona Electric 2019 but the HKG_community comes up as Mock

Sample route: 6d556f411feddcac|2020-06-25--18-50-23--0

davidew98, 2020-06-29:
I used the HKG_community build after the elect_ems commint @crwusiz talked about above and it still did not detect the car and leave dashcam mode and stayed in Mock.

Sample HKG_community route detected as Mock: 6d556f411feddcac|2020-06-29--04-16-27

Sample release2 route detected as Hyundai Kona EV and tried to engage OP but Seatbelt error, Sensor error, and CAN error: 6d556f411feddcac|2020-06-26--21-57-52

crwusiz, 2020-06-29:
> I used the HKG_community build after the elect_ems commint @crwusiz talked about above and it still did not detect the car and leave dashcam mode and stayed in Mock.
>
> Sample HKG_community route [...]

i think..the reason....is (no SCC) -> radar signal not exist...

adeebshihadeh, 2020-06-30:
@swells-Kona @crwusiz Seems like the Kona EV uses a different message for the gas pedal like the Ioniq. Fixed in 83af35bf95e794f0968fc3a8e18333d96c786172

@davidew98 cars without SCC are currently not supported.

closed 2020-06-30 by adeebshihadeh

davidew98, 2020-07-01:
> i think..the reason....is (no SCC) -> radar signal not exist...

this does not explain why release2 detects my car and HKG does not.

adeebshihadeh, 2020-07-01:
@davidew98 This isn’t the place for discussion of community forks. For that discussion, try discord.comma.ai or the issues of the specific repository you’re running code from.