commaai/openpilot #1754: git checkout is failing for this repo, please sort this!

ruddhisundar, 2020-06-21:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

It should checkout properly.

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

ruddhisundar, 2020-06-21:
$ git clone https://github.com/commaai/openpilot.git                                                                                                                                             [14:29:33]
Cloning into 'openpilot'...
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 24165 (delta 0), reused 0 (delta 0), pack-reused 24164
Receiving objects: 100% (24165/24165), 1004.47 MiB | 6.08 MiB/s, done.
Resolving deltas: 100% (13003/13003), done.
git-lfs filter-process: git-lfs: command not found
fatal: the remote end hung up unexpectedly
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

FAIL: 128

jyoung8607, 2020-06-21:
> $ git clone https://github.com/commaai/openpilot.git [14:29:33]
...
> git-lfs filter-process: git-lfs: command not found

You'll need [Git Large File Storage (LFS)](https://git-lfs.github.com/) installed in order to checkout master.

closed 2020-06-21 by jyoung8607

ruddhisundar, 2020-06-22:
Okay, sorry. I did not know something like that existed. Maybe a note in the README.MD file will help. Thank you for enlightening.

ruddhisundar, 2020-06-22:
Thank you!

Sincerely,

Rudhisundar Beura

*Karama pradhana Biswa Rachi Rakhaa | Jo jas karahi So Tas phal Chakhaa
||Hoi Hai soi jo Raama Rachi Rakha |*
*Ko Kari tarak badhavahin Sakha ||   ...Tulsi Das's magnum opus:
RamaCharitManasa, Bala Kanda   *