commaai/openpilot #1755: 2017 Lexus RX350 F Sport does not recognize C2 when installed. No calibration.

rx3123, 2020-06-21:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

rx3123, 2020-06-21:
Describe the bug
Installed Openpilot on a new Comma 2 that arrived yesterday. Car does not recognize the device.

How to reproduce or log data
Used a new install for openpilot.  Drove a few miles, the software is openpilot, but C2 seems to function only as a dashcam

Expected behavior
The website says it is compatible. I think it should recognize the car. But it does not.

Device/Version information (please complete the following information):

Device: [Comma Two]
Version: [0.7.6.1],
Car make/model [Lexus RX 350 F Sport]

VirtuallyChris, 2020-06-22:
What's your dongle id? Found in settings -> device -> dongle id

rx3123, 2020-06-22:
Hi! I will look when I get home. Thank you so much for your help!

rx3123, 2020-06-23:
Hi.
Dongle ID  3f584d02a9803d68

BTW, watching your OpenpilotYoutube videos is the reason I bought the Comma
2.

rx3123, 2020-06-23:
Route:
3f584d02a9803d68|2020-06-20--16-47-50        0.4633 miles
3f584d02a9803d68|2020-06-20--16-40-10        3.21603 miles

rx3123, 2020-06-23:
Please let me know if you need more routes with more miles driven. Thanks again!

VirtuallyChris, 2020-06-23:
Your car will now work as of this commit! This will go out in the next version of openpilot or you can switch to master-ci later today to give it a try.

https://github.com/commaai/openpilot/commit/a5ce72465b3ff7111e18433fe40e5538b85ade83

closed 2020-06-23 by VirtuallyChris

rx3123, 2020-06-23:
> a5ce724

Awesome. Thank you. Keep up the awesome videos!

rx3123, 2020-06-24:
Hi. When I tried open pilot today, it behaved the same as before. Do I have to wait until the next version of open pilot is released to have it functional? How do I switch to master ci? What is master ci?

rx3123, 2020-06-24:
"This will go out in the next version of openpilot "....does that mean I
will need to wait until the next version is released for my car to
recognize the Comma 2?  If so, when would the release usually happen?
Thanks.