commaai/openpilot #1757: No steering wheel on Ioniq

ceicomcloud, 2020-06-22:
**Describe the bug**
No steering wheel after openpiot engage
Hyundai Ioniq 2017 Hybrid with LKAS...

**How to reproduce or log data**
I have try CAR.Ioniq (in Hyundai_IONIQ Branch) and CAR.IONIQ_EV_LTD (Master Branch) in values.py and two has same.

**Expected behavior**

The car don't turn the wheel after openpilot has engage. (OP is green, no error... Speed is correct...)

**Device/Version information (please complete the following information):**
 - Device: Freon with Open Pilot 0.8 - Master
 - Dongle ID: https://my.comma.ai/useradmin/?onebox=108108934345574338940
 - Route: All route in my account
 - Version: 0.8 Master Branch 
 - Car make/model: Hyundai Ioniq 2017 Hybrid

**Additional context**

ceicomcloud, 2020-06-22:
PS: Link of route with disable OpenPilot for view the car event:
https://my.comma.ai/cabana/?route=93b03f7dcb4f0e45%7C2020-06-17--06-30-23&exp=1623912288&sig=Tyzg0Imxw2i0FE5fUffB5LwLyk3NEuvKRMzazjNq1HI%3D&max=6&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F93b03f7dcb4f0e45%2F6c4dbfa54bf8d0fede8329d34e34f052_2020-06-17--06-30-23

I have let the stock lkas... at: 80,111 and 230

adeebshihadeh, 2020-06-22:
It seems like the Ioniq Electric and Ioniq Hybrid are two different models. openpilot currently supports the Ioniq Hybrid. New models require car ports from the community. Did you have to make any code changes to get openpilot to engage?

ceicomcloud, 2020-06-23:
No code change for CAR.IONIQ and car CAR.IONIQ_EV_LTD for engage OP

s70160, 2020-06-23:
maybe you can use HGK7.3_KEGMAN it's work,you can search some video on yourtube!

adeebshihadeh, 2020-06-23:
The Ioniq Hybrid is a different model, so it will need a new car port. See the  [README](https://github.com/commaai/openpilot#community-maintained-cars-and-features) for the list of currently supported cars.

closed 2020-06-23 by adeebshihadeh