commaai/openpilot #1765: Stops communicating with 2017 Corolla until EON reboot

hansonr55, 2020-06-23:
**Describe the bug**

With the EON powered up, turning car on will show Vehicle Connected but will not communicate until EON reboot.  After reboot, it will work as intended.  If car is turned off and back on, communication will stop until EON is rebooted again.

**How to reproduce or log data**

**Expected behavior**

Continued communication as worked with 0.7.4

**Device/Version information (please complete the following information):**
 - Device: EON
 - Dongle ID: ce0ea5158a0e1080
 - Route: Any drive after 0.7.6 was loaded
 - Version: 0.7.6
 - Car make/model: Toyota Corolla 2017

**Additional context**

pd0wm, 2020-06-23:
I'm not sure what you mean with "stops communicating". Is it not going onroad (showing the camera view) or is it not recognizing your car?

hansonr55, 2020-06-23:
It's acting as if it's not recognizing the car.  OP won't engage, dash has all errors as if EON wasn't connected at all yet EON shows vehicle connected.

If I restart the EON, even while I'm driving, it's all good until I turn the car off again.

Ryan

pd0wm, 2020-06-23:
Is it showing the camera view when this happens? Or the screen with your drive statistics?

Can you link to a specific drive where you had to restart?

hansonr55, 2020-06-23:
My drive from this morning at around 7:30 Central, I had to restart the EON before I could engage OP.
Any drive since loading 0.7.6 requires EON reboot.

Errors on dash - headlight, lane departure alert, front camera.

EON shows the camera view but won't engage.

Thanks,

Ryan

hansonr55, 2020-06-23:
I may have figured the issue out. I noticed I had lots of storage used and that I kept getting the connect to internet alert. I think that even though it showed “connect online” that it couldn’t communicate with comma.ai. Possibly a firewall issue.

When I paired with my cell phone the error went away and it worked as intended.

Thanks,

Ryan

closed 2020-06-24 by VirtuallyChris