commaai/openpilot #1767: "Warning: No CARLA" error

arsalhuda24, 2020-06-23:
when I run ./bridge file as explained in README.md it throws the following error and fails to import carla and raises the following exception. The ./start_carla.sh though runs fine and I can see the carla UI. 

ECU Camera Simulated: True
ECU Gas Interceptor: True
fatal: HEAD does not point to a branch
tools/sim/start_carla.sh: needs update
WARNING: NO CARLA

I am using python3.7.3. 

Really appreciate if you can help on this.

sjdavalle, 2020-06-24:
I would suggest running ubuntu_setup.sh first order to get Pyenv properly installed... 
it will also install Python3.8

arsalhuda24, 2020-06-25:
Thanks sjdavalle. I ran ubuntu_setup.sh and i dont get import carla error instead now i get the following:

ECU Camera Simulated: True
ECU Gas Interceptor: True
fatal: HEAD does not point to a branch
tools/sim/start_carla.sh: needs update
Traceback (most recent call last):
  File "./bridge.py", line 170, in <module>
    go()
  File "./bridge.py", line 121, in go
    imu_bp = blueprint_library.find('sensor.other.imu')
IndexError: blueprint 'sensor.other.imu' not found
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 186 (sensor.camera.rgb) 
terminate called after throwing an instance of 'boost::python::error_already_set'
Aborted (core dumped)

adeebshihadeh, 2020-06-25:
#openpilot-simulation on the [community discord](https://discord.comma.ai) is a better place to discuss getting CARLA installed and setup.

closed 2020-06-25 by adeebshihadeh