commaai/openpilot #1777: Allow gas press to resume

adeebshihadeh, 2020-06-25:


cydia2020, 2020-06-27:
+1 on this.

Scottgagnebin, 2020-07-11:
+2 on this -- 2018 Camry, would be nice to help the car along in traffic too

shazooer, 2020-09-16:
+3 on this.

VirtuallyChris, 2020-09-24:
I had a user let me know about how this feature would help make their openpilot experience better.

```(2018 Prius Prime) uses the accelerator as the primary method of resuming cruise control after coming to a full stop.  The only other way to resume cruise control after coming to a full stop is frantically pressing up on the cruise control knob, but this also increases the set speed which isn't wanted.```

Scottgagnebin, 2020-09-25:
@VirtuallyChris This is the same for the 2018 Camry. Often my cruise control is set to 115 in stop and go traffic. Fun times

4oSmile, 2020-11-15:
+4 on this 
When ACC slows down and full stops in traffic - gas press is most uses to go again BUT this disengage the OP and I need again to resume it from the cruise lever or button on newer models, so please "DO NOT DISENGAGE OP on GAS PRESS" or add is as option in OP settings (Prius 2016)

cydia2020, 2020-11-15:
> "DO NOT DISENGAGE OP on GAS PRESS" or add is as option in OP settings (Prius 2016)

Likely won't happen.

What I can think of is to let the driver press the accelerator pedal ONCE when the vehicle is at standstill. And once the vehicle reaches 15km/h accelerator pedal will disengage OP.

Scottgagnebin, 2020-11-15:
At 15mph is awesome as well. Recommended a setting to enable only on those
where OP can't control. Safety

4oSmile, 2020-11-15:
> > "DO NOT DISENGAGE OP on GAS PRESS" or add is as option in OP settings (Prius 2016)
> 
> Likely won't happen.
> 
> What I can think of is to let the driver press the accelerator pedal ONCE when the vehicle is at standstill. And once the vehicle reaches 15km/h accelerator pedal will disengage OP.

Ok, I update me opinion  **WHEN THE CAR IS STOPPED**, I didn't men when car is moving. 
But I catch the point because pressing accellerator pedal, the car starts moving, so it must be some time/speed for depress (during pedal is still pressed and car is moving), so 5, 10 or 15 km/h is needed.

4oSmile, 2021-05-18:
Half year later...

Any thoughts of including such an option when the car is completely stopped?

(Of course, in the situation the OP must not be switched off immediately, the car starts at 1,2,3,4,5 km/h before the reaction from the release of the accelerator pedal, for example on time or speed)

And at the moment when the car is completely stopped and we press the button/lever for "RESUME" - the car starts and the stock ACC of the car takes over the its functions. It does not goes if there is an obstacle in front!

By lightly depressing the accelerator pedal, the same thing happens with the stock dynamic cruise control, simply the OP is switched off and you need to "RESUME" the OP once more from the button/lever. As all this is accompanied by the sounds of stopping and starting OP.

And at the moment when the car is completely stopped and we press the button / lever for "recovery" - the car starts and the ass of the car takes over the functions. It does not start if there is an obstacle in front!

With a light press of the gas, the same thing happens with the standard dynamic cruise control, just the OP is turned off and you need to run the OP once more from the button / lever. As all this is accompanied by the sounds of stopping and starting.

jedberg, 2021-06-21:
I would like this functionality when the car is moving too.  On my stock Honda AC, I can press the gas and it doesn't disengage.  This is especially important on hills, when the car doesn't press the gas hard enough on its own.

lskov, 2021-09-05:
I see the point of this function, but I also see it’s maybe not should be relevant if the parameters for the car is tuned well. Like discussed on this thread [21697](https://github.com/commaai/openpilot/issues/21697)
Personally I would prefer the parameters for the car is tuned properly so it’s not necessary to press the pedal.

adeebshihadeh, 2021-12-15:
Closing this as it's not something we're working on anytime soon. We'll likely have to look into this once we do stop signs and red lights.

closed 2021-12-15 by adeebshihadeh