commaai/openpilot #1793: lka_state often gets faulty. (error values are 9 and 17)

clepz, 2020-06-27:
**Describe the bug**
My toyoto corolla hybrid sends 9 and 17 for lka_state and then disable (value is 1) very often when I use joystick debug tools. I can't drive with joystick because of that.
 
If lka sends error that value is 17, I have to restart the car to get control again. 

I tried to change steerlimitparams but result is the same. I tried to understand when the errors were produced but I couldn't find any pattern.

**Expected behavior**
Value of lka_state that comes from car should always be 5.

Operating system: Ubuntu 16.04

adeebshihadeh, 2020-06-27:
The joystick is not meant for driving. Does this happen in normal operation with openpilot?

clepz, 2020-06-27:
I know that joystick not meant for driving but I just want to understand that why I get these errors, what is wrong with my car? BTW My speed was 5-10 kmph while testing. 

I didn't encounter with these problems with normal openpilot but I didn't check the lka_state(I mean open pilot seemed normal ) .I'll go for a drive and check the value of lka asap and I'll share.

clepz, 2020-06-30:
if I use safety_alloutput, I can't control gas and set cruise . But if i use safety_toyota and set cruise then everything is ok with gas and break.
I'll try the same solution with steering. My laptop is out of charge after gas test :smile:

adeebshihadeh, 2020-08-16:
Safety all output won't work since it doesn't handle any of the forwarding.

closed 2020-08-16 by adeebshihadeh