commaai/openpilot #1799: OpenPilot Tools Does Not Work Following the readme

mandray, 2020-06-30:
**Describe the bug**

I've been trying to get the joystick example under tools to work for about a week now. I feel like there are things missing from the readme, it definitely doesn't work right out of the box.
After following all the steps, Running Boardd without sudo gives me `Permission Denied` for one

Then running it with sudo does show that the panda is found and the correct panda with gps is detected, but it's aborted right after
```
Aborted (core dumped)
```

This is all with the car on and my paw plugged into the panda. I tried the paw switched to both off and PWR
If the car is off (on idle) it does seem to show it on, but since the car is off I wouldn't be able to do anything
Shows
```
Panda GPS on
```

debug_controls.py doesn't seem to start either. It throws this error
```Could not open /dev/shm/can```

I'm not sure what to do at this point, but I feel like i've tried everything.

Beforehand I did run the udev tools

**How to reproduce or log data**

Follow the readme for tools on a fresh install of Ubuntu 16.04 with a grey panda

**Expected behavior**

Joystickd to work after following the readme with a Panda Paw

**Additional context**

I use a Panda Paw, I tried using the paw with both the Pwr switch on and off, but neither worked

Operating system: Ubuntu 16.04

codename224, 2020-07-04:
I agree, the readme should mention things like udev rules for panda, flashing the panda on a PC and not using the eon panda firmware, safety modes, controls allowed, switches on giraffe hardware (even though most of us have harness now), and other gotchas that come up during the process. It is not as easy as running the three commands in the readme at all. I have been trying to get it to work for a few months now off and on with no luck.

roxasthenobody98, 2020-07-27:
I have noticed that with the /dev/shm/can and /dev/shm/sendcan, if you do sudo rm -rf /dev/shm/can && rm -rf /dev/shm/sendcan it can latch onto those files

adeebshihadeh, 2021-05-21:
Made a new issue to track this #1799.

closed 2021-05-21 by adeebshihadeh