commaai/openpilot #1814: Support lexus rx300 2019?

daniel1975-sys, 2020-07-04:
Today i tested it but it does not work.
Why? 

[image]

adeebshihadeh, 2020-07-04:
It seems like there's a discrepancy between the [README](https://github.com/commaai/openpilot#supported-cars) and the shop, but it should be straightforward to add support. Can you send me your [dongle ID](https://github.com/commaai/openpilot#supported-cars)? Our community discord (https://discord.comma.ai) can also help you get setup and running.

daniel1975-sys, 2020-07-05:
I do not know where i can find dongle ID.
I attached order #8795

Can you explain more how to find dongle id?

adeebshihadeh, 2020-07-06:
The instructions are on the wiki: https://github.com/commaai/openpilot/wiki/FAQ#what-is-a-dongle-id

VirtuallyChris, 2020-07-07:
We've pulled your firmware versions and added it to a test branch. Can you let us know if this works for you?
[https://github.com/commaai/openpilot/tree/lexus_rx_test](https://github.com/commaai/openpilot/tree/lexus_rx_test)

adeebshihadeh, 2020-07-10:
Closing the issue since the discrepancy was fixed on the website. Contact support if you’d like to return the product, otherwise a pull request was opened to add support for your car.

closed 2020-07-10 by adeebshihadeh

daniel1975-sys, 2020-07-13:
My doggle id

5df2c1f37d85169e

I drove yesterday for 30 min for shopping.
when I activate cruise control, I did not see handle picture on the screen.

do i need to download or install something?
or do i need wifie connection only?

cheers,
daniel

VirtuallyChris, 2020-07-15:
Hi Daniel,

you must switch to the branch we setup for you and let us know if it works https://github.com/commaai/openpilot/tree/lexus_rx_test

daniel1975-sys, 2020-07-16:
you mean i need to type
https://github.com/commaai/openpilot/tree/lexus_rx_test
for enter the ULL of the program to install?

right?
[image: Screen Shot 2020-07-16 at 1.11.13 PM.png]

adeebshihadeh, 2020-07-16:
@daniel1975-sys No, there's a wiki guide here showing the steps: https://github.com/commaai/openpilot/wiki/UpdateModifyOpenpilot. Alternatively, our community discord (https://discord.comma.ai) can also help you switch to the test branch.

daniel1975-sys, 2023-08-05:
> UpdateModifyOpenpilot. Alternatively, our community discord (

Hi
Finally i saw you upgrade for lexus 2019
But i found comma does not support comma 2 anymore.
Do i need comma 3 for lexus 2019?

I bought comma 2 in 2020 during covid pandemic.
I hoped comma upgrade it for lexus 2019.
Can you give me comma 3 ?

Daniel

PR #1838 (2020-07-27): Add Lexus RX300 2019 FW
resolves #1814