commaai/openpilot #1818: Need help

DeeSpeed, 2020-07-05:
I have been trying for weeks to get questions answered to no avail.  I have tried Discord (this is a gamers site and I have no idea why this company would use it) and other avenues.  If anyone has suggestions I would appreciate it.  I am sure this is not the proper place to post this but I am very interested in this device.  Thank you.

TK211X, 2020-07-05:
What vehicle is this for?

What questions do you have?

Are you capable of coding?

Comma decided to switch to Discord because it's free. They expect capable people to join and help themselves if an answer to the question they have exists elsewhere.

[image]

TK211X, 2020-07-05:
See:

https://www.youtube.com/watch?v=jxg1KuowODA

closed 2020-07-05 by adeebshihadeh