commaai/openpilot #1829: OpenPilot goes on-road before checking thermal limits on device startup.

Hubblesphere, 2020-07-06:
**Describe the bug**

When starting car with a shutdown device OpenPilot goes onroad first, then checks thermal limits. If already overheated it will switch back offroad causing dash errors. 

**How to reproduce or log data**

Start car with Comma Two in power save state while in a high temperature environment. 

**Expected behavior**

Device will not go on-road due to already high thermal limit.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: f91c796ac0bf5eb6
 - Route: unavailable
 - Version: master-ci
 - Car make/model: 2019 Toyota Corolla Hatch

pd0wm, 2020-07-06:
This is not exactly what is going on. The problem is that openpilot goes onroad, starts using more CPU, and then quickly heats up. After it heats up it can go back offroad. There is already a pretty big margin before it allows you to offroad, but I increased it some more today. Let me know if you still have problems.

The current temperature limits are: go onroad only below 70 C max cpu temperature, go offroad above 107 C max cpu temperature.

Hubblesphere, 2020-07-06:
Unfortunately explorer didn't record the event for some reason but at startup Comma Two was reading 76C yet still went on-road before switching off-road. Could have been right at old limit and quickly ramped up since it was coming from a powered off state before display could actually show temp.

pd0wm, 2020-07-06:
The ui shows a slightly different temperature, but CPU temperatures are usually slightly higher. But I assume that was before my change?

Hubblesphere, 2020-07-06:
Yes before your latest change. We can close this and I will report if it's experienced again!

closed 2020-07-06 by Hubblesphere