commaai/openpilot #1859: Hide camera view when in reverse

adeebshihadeh, 2020-07-10:


nabeelr, 2020-08-11:
This is a severely under-rated feature request.

Especially when mounted close to the rearview mirror, because it's brighter than the rearview mirror, it tends to catch your eye, and it takes specific intention to avoid looking at it and instead use the rearview mirror as intended.

Lots of cognitive load to ignore it, especially when it's dark. I've caught myself staring at it blankly not understanding why I'm not seeing what I expect and the mistake I made. This was especially bad after I first installed it.

The screen going black when reversing will help minimize distractions when preforming delicate operations.

getalex, 2020-09-01:
As opposed to a global feature I propose this being a setting: "Fade screen during reverse" or some other verbiage.. 
For dashcam purposes the feed should still be capturing but the screen may dim/fade/disable during reverse via this option.

PR #2144 (2020-09-09): hide camera view when in reverse
fixes #1859
closed 2020-09-09 by adeebshihadeh