commaai/openpilot #1868: Error; Driver model lagged

adam-nelson, 2020-07-12:
**Describe the bug**

Engaging openpilot causes master error driver model laggged

**How to reproduce or log data**

Engage openpilot and wait 30 seconds before error appear

**Expected behavior**

N/A

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: 18e8ac91b805db11
 - Route: 18e8ac91b805db11|2020-07-11--18-36-12--354
 - Version: master-ci 0.7.7-release
 - Car make/model:Toyota corolla Hybrid 2020 hatchback Right hand drive

**Additional context**

Error happen pretty consistently every 10 seconds before i disengaged. Testing the master-ci branch because of an another error

adam-nelson, 2020-07-12:
Just checking if this is new bug or old bug?

xps-genesis, 2020-07-13:
i have had similar experience during a recent short trip. only with on a reboot of comma, it made it happy for the rest of the trip or the alert would come one consistently for every 5-10seconds of op engagement

pd0wm, 2020-07-15:
Should be fixed by https://github.com/commaai/openpilot/commit/58c3b5ba84f6c3da3c660770148cea98c708fce5. Feel free to reopen an issue if you still get too much of these warnings.

closed 2020-07-15 by pd0wm