commaai/openpilot #1876: openpilot can enable at startup while thermal status too high

Hubblesphere, 2020-07-13:
*description of bug*
At initial on-road from startup or after cooling from thermal on-road protection open pilot can enable for first ~25-30 seconds after going on road. (tested in master-ci) It will then alert and disable after 20-30 seconds of being engaged. 

**How to reproduce or log data**

Enable openpilot within ~25-30 seconds of going on-road while temps in yellow or above.

**Expected behavior**

openpilot should not allow enable

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: f91c796ac0bf5eb6
 - Route: f91c796ac0bf5eb6|2020-07-13--12-02-21--0
 - Version: commit 7ac0040
 - Car make/model: 2019 Toyota Corolla Hatchback

**Additional context**

Alternate Route example: f91c796ac0bf5eb6|2020-07-08--12-08-59--0

techasit14607, 2020-07-25:
Fix bug.

TK211X, 2020-08-16:
Could we also close the switch so that vehicle can be stock of temp is too high. Currently it lights up the dash and is very annoying.

adeebshihadeh, 2020-11-16:
This should be fixed now, especially with the recent thermal improvements due to the CPU usage reduction and downclock.

closed 2020-11-16 by adeebshihadeh