commaai/openpilot #1906: [GM] Cannot resume after stop

kishkaru, 2020-07-22:
**Describe the bug**

#1729 added a change to show an alert (`EventName.belowSteerSpeed`) when `vEgo < minSteerSpeed`. That change finally got merged into a release (0.7.7) this week and I got to test it out. There is a potential unintended consequence of that change:

1) The alert when stopped that used to show "Stopped. Press resume to resume" is no longer showing when stopped.
2) Pressing resume from a stop doesn't do anything
3) `belowSteerSpeed` alert is always displayed when `vEgo < minSteerSpeed`, regardless of stopped status.

**How to reproduce or log data**

Come to a complete stop. Press resume.

**Expected behavior**

1) "Stopped. Press resume to resume" alert should display.
2) Pressing resume should resume.

**Device/Version information (please complete the following information):**
 - Device: EON
 - Dongle ID: N/A
 - Route: N/A
 - Version: 0.7.7
 - Car make/model: Chevy Volt

**Additional context**

Maybe we should change it to show the "Stopped" alert when at 0mph, instead of the `belowSteerSpeed` message.
IMO, it doesn't make sense to display the "steer unavailable" message at 0mph anyway, since we don't need to steer when stopped.

techasit14607, 2020-07-25:
Fix bug

kishkaru, 2020-08-01:
I removed that new alert in 0.7.7, and the `Stopped. Press resume to move` alert reappeared once at stopped. The resume alert disappears when pressing resume, but the actual resume is not working.

I went back and tried out 0.7.6.1, and resume is also not working there. I kept going back, and its broken in: 0.7.5, 0.7.4, 0.7.3.
Unfortunately, 0.7.2 and earlier won't load on my EON, maybe because NEOS versions changed:
```
/data/openpilot/launch_chffrplus.sh: line 61: echo: write error: Permission denied
Traceback (most recent call last):
  File "./manager.py", line 12, in <module>
    from common.basedir import BASEDIR, PARAMS
  File "/data/openpilot/common/basedir.py", line 4, in <module>
    from common.android import ANDROID
  File "/data/openpilot/common/android.py", line 8, in <module>
    from cereal import log
ImportError: cannot import name 'log' from 'cereal' (unknown location)
```
So it's unclear which version of OP broke it. Maybe 0.7.1? `Improve low speed ACC tuning` Or 0.7.0? `New driving model trained end-to-end: improve lane lines and lead detection`.

adeebshihadeh, 2020-12-19:
Fixed in #2709

closed 2020-12-19 by adeebshihadeh