commaai/openpilot #1920: Dongle ID 5b49a2b01b14aa75 - serial# fa70ca4d-2017 Prius prime

cheta48, 2020-07-24:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

front camera not working.

techasit14607, 2020-07-25:
Help fix it.

techasit14607, 2020-07-25:
Open camera.

cheta48, 2020-07-25:
Only can see the menu/setting. No pic of road

adeebshihadeh, 2020-08-02:
Closing due to insufficient information. Feel free to reopen after filling in all the information in the bug report template.

closed 2020-08-02 by adeebshihadeh