commaai/openpilot #1929: Add missing Hyundai test routes

adeebshihadeh, 2020-07-26:
Each route should be at a least a few minutes with stock openpilot engaged.

* [ ] HYUNDAI.HYUNDAI_GENESIS
* [ ] HYUNDAI.ELANTRA
* [ ] HYUNDAI.ELANTRA_GT_I30
* [ ] HYUNDAI.GENESIS_G80
* [ ] HYUNDAI.GENESIS_G90
* [ ] HYUNDAI.KIA_FORTE
* [ ] HYUNDAI.KIA_OPTIMA
* [ ] HYUNDAI.KIA_OPTIMA_H
* [ ] HYUNDAI.KIA_SORENTO
* [ ] HYUNDAI.KIA_STINGER
* [ ] HYUNDAI.KONA
* [ ] HYUNDAI.KONA_EV

xps-genesis, 2020-07-27:
Add these to list

HYUNDAI.KONA_HEV,
HYUNDAI.GRANDEUR,
HYUNDAI.GRANDEUR_HEV,
HYUNDAI.KIA_NIRO_HEV,
HYUNDAI.SONATA_H,
HYUNDAI.KIA_CADENZA,
HYUNDAI.KIA_CEED,
HYUNDAI.KIA_CADENZA_HEV,
HYUNDAI.KIA_SPORTAGE,

adeebshihadeh, 2020-07-27:
Those don’t need to be added since we don’t currently support them. They simply won’t be merged in until we have a test route.

xps-genesis, 2020-07-27:
i am a little confused, because people will not run the default stock release from comma since the vehicles are not supported. so we may find it even more difficult to get routes of the new vehicles running stock.

let me know what you think.

adeebshihadeh, 2020-07-27:
Before any new model gets merged in, someone with the car has to test the branch that we intend to merge in. The route they used to test it can serve as the test route.

xps-genesis, 2020-08-13:
KONA and OPTIMA route details added

adeebshihadeh, 2020-09-13:
Moved to #978

closed 2020-09-13 by adeebshihadeh