commaai/openpilot #1943: Problem loading dmonitoring_model.keras

weiwang928, 2020-07-28:
Ubuntu 18.04
Python 3.6.9
Tensorflow version = 2.0
`from tensorflow.keras.models import load_mode
`
`
model = load_model('dmonitoring_model.keras')
`

<br>
Error:

  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/saving/save.py", line 146, in load_model
    return hdf5_format.load_model_from_hdf5(filepath, custom_objects, compile)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/saving/hdf5_format.py", line 168, in load_model_from_hdf5
    custom_objects=custom_objects)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/saving/model_config.py", line 55, in model_from_config
    return deserialize(config, custom_objects=custom_objects)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/layers/serialization.py", line 102, in deserialize
    printable_module_name='layer')
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/utils/generic_utils.py", line 191, in deserialize_keras_object
    list(custom_objects.items())))
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/engine/network.py", line 906, in from_config
    config, custom_objects)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/engine/network.py", line 1842, in reconstruct_from_config
    process_layer(layer_data)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/engine/network.py", line 1824, in process_layer
    layer = deserialize_layer(layer_data, custom_objects=custom_objects)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/layers/serialization.py", line 102, in deserialize
    printable_module_name='layer')
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/utils/generic_utils.py", line 191, in deserialize_keras_object
    list(custom_objects.items())))
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/layers/core.py", line 862, in from_config
    config, custom_objects, 'function', 'module', 'function_type')
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/layers/core.py", line 914, in _parse_function_from_config
    config[func_attr_name], globs=globs)
  File "/home/wei/.local/lib/python3.6/site-packages/tensorflow_core/python/keras/utils/generic_utils.py", line 285, in func_load
    code = marshal.loads(raw_code)
ValueError: bad marshal data (unknown type code)

gucasbrg, 2020-07-28:
the same question

ZwX1616, 2020-07-29:
Could you try loading the model in TF2.2?

weiwang928, 2020-07-30:
I tried both TF2.1 and TF2.2 as well, but I got this error:

TypeError: Error converting shape to a TensorShape: Dimension value must be integer or None or have an index method, got 80.0.

ZwX1616, 2020-07-30:
> I tried both TF2.1 and TF2.2 as well, but I got this error:
> 
> TypeError: Error converting shape to a TensorShape: Dimension value must be integer or None or have an index method, got 80.0.

Ah my bad. Try python2.7 and TF1.14 this time

weiwang928, 2020-07-31:
Works now with python2.7 and TF1.14. Thanks!

closed 2020-07-31 by weiwang928