commaai/openpilot #1976: shutil 'module' object has no attribute 'which'

mnsh0409, 2020-08-04:
When I ran scons -j4, it shows the following error:

scons: Reading SConscript files ...
AttributeError: 'module' object has no attribute 'which':
  File "/home/martin/openpilot/SConstruct", line 221:
    SConscript(['cereal/SConscript'])
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 614:
    return method(*args, **kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 551:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/scons/SCons/Script/SConscript.py", line 260:
    exec _file_ in call_stack[-1].globals
  File "/home/martin/openpilot/cereal/SConscript", line 14:
    if shutil.which('capnpc-java'):

Operating system: [Ubuntu 16.04  Python 3.7.3]

adeebshihadeh, 2020-08-04:
`shutil.which()` has been part of python since 3.3. Make sure you're actually running python3.

closed 2020-08-04 by adeebshihadeh

mnsh0409, 2020-08-05:
I has already used Python 3.7.3 in conda

pd0wm, 2020-08-20:
Make sure you install scons using pip instead of using the ubuntu supplied one.