commaai/openpilot #1980: Openpilot unable to update even with an Internet connection, eventually won't engage

cb750comma, 2020-08-05:
**Describe the bug:**
Openpilot unable to update even though it is connected to the internet, eventually Openpilot won’t engage completely.
I received the message "Immediately connect to the internet... openpilot won't engage in 0 days" and eventually a message to say “openpilot won’t engage until it connects to the internet....”   I have confirmed there has been a good internet connection.for many hours
The C2 shows as connected I have tried multiple reboots. I have tried connecting to different SSIDs on both the 2.4 ghz and 5.0 ghz band. Brought it inside my house to connect, left it connected for hours. 

**How to reproduce or log data**
See the “mount” screenshots in this link
https://discordapp.com/channels/469524606043160576/524592892627517450/739335878320586782

**Expected behavior**
The device show itself as online, should automatically perform the update and not lock out openpilot from engaging

**Device/Version information**
Device: Comma 2
Dongle ID: 3c7e4415c21fbe0b
Version: 0.7.7  (no custom changes)
Car make/model: Hyundai Palisade 2020

**Additional context**

Community member: Nelsonjchen spent  time trying to diagnose this issue and below are his comments

https://discord.com/channels/469524606043160576/621451618122072123/740314745357336687

“just post the screenshots you took and mention/attach the mount thing or screenshots which is peculiar. i didn't do much diagnosing other than that. You can also link the discord message where the conversation happened. Beyond that, I have no idea how to diagnose the updater. it's probably something fatal early before it even logs since adeeb didn't see any updater logs. How to debug after that, I don't know. here's the discord link to the odd mount screenshot thing you can use in your report: https://discordapp.com/channels/469524606043160576/524592892627517450/739335878320586782 that said, the mount thing is peculiar, but it could be a symptom and not a cause or possibly not even related. anyway, make the bug report, and hopefully comma might be able to advise on how to debug the updater.”

tarasglek, 2020-08-09:
0.7.7 didn't work for me. I went back to 0.7.6.1 git branch, after a week I started getting getting same error message as above.

[image]

adeebshihadeh, 2020-08-10:
@tarasglek in your case, the updater is failing to pull from your fork which you cloned using SSH, but there's no SSH key on your device. the default git key path is `/data/gitkey`

gregjhogan, 2020-08-10:
improvement for this situation mentioned here: https://github.com/commaai/openpilot/issues/1073

nelsonjchen, 2020-08-10:
@cb750comma 

https://discordapp.com/channels/469524606043160576/524327905937850394/742170315748016229

If you haven't done the uninstall and reinstall, we can try gathering more information with this.

4oSmile, 2020-08-10:
I have same problem.

Starts yesterday. 

Now even no camera is displayed. 
Only status screen with date, release, past week and all time. 

[image]

Vehicle online, connect online, online via app on phone. 

Even try uninstall and install OpenPolot again.

Never install any forks or clone ids or whatever. 

[image]

What is the problem? 
What happens? 
I have been using it for three weeks.

cb750comma, 2020-08-10:
Nelson,  I have not done the re-install yet.  We can do more information
gathering if you want to do it tonight at 10:30pm EST (7:30 pm Phoenix
time) or if you prefer 11:00pm EST.
You can PM me directly.  Let me know what time you prefer.

Thanks, Harry

4oSmile, 2020-08-10:
Good news from me:
After reinstall of OpenPilot, as I said same status scrrrn only, after that made reset calibration, still only status screen, I leave the device in working car for ~15 minutes to cool down temperature inside the car and when I came back it was no Calibration screen 0% and camera working, drive half km and now everything works again.

Also it might be because someome (in comma team) fix something and not because of reinstallaton.

nelsonjchen, 2020-08-10:
> Nelson, I have not done the re-install yet. We can do more information gathering if you want to do it tonight at 10:30pm EST (7:30 pm Phoenix time) or if you prefer 11:00pm EST. You can PM me directly. Let me know what time you prefer. Thanks, Harry
> […](#)

I'll hit you up on Discord at 10:30pm EST tonight.

adeebshihadeh, 2020-08-11:
This was an issue with @cb750comma's network.

closed 2020-08-11 by adeebshihadeh

nelsonjchen, 2020-08-11:
Here's what we discovered:

ICMP out or in doesn't work on @cb750comma 's network. Context:

https://discordapp.com/channels/469524606043160576/524592892627517450/742578900869185576

Without ping working, the updater will not proceed: 

https://github.com/commaai/openpilot/blob/c2a6ae6e540e2783ff17034a5cf8f62e5d275ebb/selfdrive/updated.py#L263

Ping failure and incorrect time are also events that are not logged when it comes to update failures. This is more for anyone looking at logs from the comma side.

cb750comma, 2020-08-12:
I would not simply say the issue here is with the network.  I have not experienced any network issues running or updating another other application.  I successfully installed Openpilot over this network.
It might be accurate to say the factory default security configuration of my network/Modem/Router is not compatible with the Openpilot Update code.
I want to make this distinction since it might help someone else that runs into this issue.  My Internet Service Provider is Rogers in Ontario and the Modem/Router is Hitron Model: CODA-4582

nelsonjchen, 2020-08-12:
In the time I was helping, I also could not figure out a way to enable ICMP out on @cb750comma 's network either. The firewall option described in the manual configures the option to block ICMP going to the router from external hosts; it does not seem to affect ICMP out. I'm at a loss at how to suggest @cb750comma go about resolving this. I wouldn't be surprised if @cb750comma might be SOL on fixing the ICMP out issue.

I'm curious if a pull request that replaces the ping command with `curl -s -o /dev/null -w "%{http_code}" http://clients3.google.com/generate_204 | grep '204'` would be accepted. This is inspired by Android/ChromiumOS's checking @ https://www.chromium.org/chromium-os/chromiumos-design-docs/network-portal-detection. Android itself probably uses a variant of this to check open wifi networks for connectivity.

This isn't just for @cb750comma , but I think it would improve that part of the code's accuracy and functionality:

* There are scenarios where ICMP ping may be allowed but HTTP or TCP or whatever isn't as in some captive portal scenarios somewhere. It is possible for an update to be attempted on these setups even though connectivity won't happen. I don't have an example off-hand but I know from the heart there's some network out there probably configured like this.
* Normal openpilot operation including uploads and updates does not require ICMP; it does require a general working connection to the internet via HTTP; this command would be a more accurate test of that.