commaai/openpilot #1996: LKAS fault: Restart car to engage

atstevenha, 2020-08-08:
**Newly installed Comma 2 for a 18 Grand Cherokee with adaptive cruise. Once installed and connections checked. The screen keeps reading LKAS fault. It seems like signal triggers or slow speeds. Please help. **

adhintz, 2020-08-08:
If you only engage OpenPilot when you're going more than 39mph do you still see the LKAS faults?

atstevenha, 2020-08-08:
> If you only engage OpenPilot when you're going more than 39mph do you still see the LKAS faults?

I can't even engage actually. Hasn't work since I've install it.

atstevenha, 2020-08-08:
[image]

cydia2020, 2020-08-08:
Seems to be the dashcam version, install openpilot during setup by choosing custom software, and type https://openpilot.comma.ai in the address bar.

closed 2020-08-09 by adeebshihadeh