commaai/openpilot #2016: Comma two not updating

haddsomnus, 2020-08-11:
I'm getting "Immediately connect to the internet to check for updates. if you don't connect to the internet openpilot won't engage in 1 day" alert even when the internet is connected. I tried rebooting but the alert still appears.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 6b5520e0ab1c7118
 - Route: 6b5520e0ab1c7118|2020-08-11--11-10-35
 - Version: 0.7.7 release2
 - Car make/model: [Hyundai santa fe 2019]

adeebshihadeh, 2020-08-11:
This is the error being throwing in the updater: `fatal: .git/index: index file smaller than expected`. Have you ever run a different branch than release2 or SSH'd into your device?

haddsomnus, 2020-08-11:
NO, I just followed the video in wiki,  pretty sure it is cleaning install no addition change or do anything different

haddsomnus, 2020-08-11:
I tried HKG before but reinstall back to release2 due to Can error. If I want to fid git index how can I do that?

closed 2020-08-15 by haddsomnus

neshant, 2021-09-14:
> I did reinstall couple time but this issue still appears

Same behavior is reproducible for me as well, I was on 0.8.7 and did a reinstall to upgrade to 0.8.8 and started getting the error again.