commaai/openpilot #2021: OpenPilot PC Simulation

mouLm, 2020-08-12:
I am hoping to validate a testing method on OpenPilot as it is one of the few opensource automotive software. The testing method relies on sending inputs to a module and observing the code coverage and output to understand the system's behavior.
 
I am targeting to perform the testing on PC and not use any hardware device. I have compiled OpenPilot successfully on Ubuntu 16.04.
 

- Ideally, I would like to send CAN messages and observe the outputs of the system. I don't have to test the full system; a small functionality can be sufficient. Is this achievable without using any hardware device?
- Is there any documentation that describes a correct behavior/output? 
- I replayed driving data using unlogger with remote data using the example python replay/unlogger.py '3533c53bb29502d1|2019-12-10--01-13-27'. I couldn't access the data that is being fetched from the route-name. How can I test the system with other inputs? From where can I get sample data so I can build similar files and experiment with different inputs?

adeebshihadeh, 2020-08-12:
We have a test called [process replay](https://github.com/commaai/openpilot/tree/master/selfdrive/test/process_replay) that seems to be what you're looking for. It runs in CI, and it's targeted to run on PC. `test_processes.py` has some sample routes that you can use.

mouLm, 2020-08-12:
Thank you for your quick reply. 
Is there a way to create/alter a route? 
If I am not mistaken, in process replay a changed code will run against the same routes and be compared with a known good output.
I am targeting testing the process on different fuzzy inputs.

adeebshihadeh, 2020-08-13:
Each process has a [config](https://github.com/commaai/openpilot/blob/master/selfdrive/test/process_replay/process_replay.py#L213) that specifies its inputs and outputs. Instead of replaying a drive, you can build your own messages using whatever inputs you want. If you want to reuse the `replay_process` function, then you'd have to build all the msgs in advance (the `lr` input is just a list of messages).

closed 2020-08-15 by adeebshihadeh