commaai/openpilot #2042: Update causes disengagement

adam-nelson, 2020-08-18:
**Describe the bug**

Since the updated method has changed in 0.7.8. 

I've have noticed a restart/reset that causes disengagement of the openpilot

**How to reproduce or log data**

Driving or not i openpilot with engage and pushing an update from github

**Expected behavior**

An update should trigger while the car is on or driving.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 18e8ac91b805db11
 - Version: 0.7.9 master-ci
 - Car make/model: Toyota corolla hatchback au delivered 2019

**Additional context**

I assume this being triggered by any github push trigger to master/master-ci branch

adeebshihadeh, 2020-08-18:
The updater doesn't run when onroad. I checked your last five routes, but didn't seen anything. Do you have a route where this happened? Has this happened more than once and why do you think it's the updater?

pd0wm, 2020-08-18:
An update also should never cause the device to reboot by itself. Only when you press the reboot button after a drive it will restart.

VirtuallyChris, 2020-08-18:
Wondering if the device is bootlooping to some extent. If it continues to reboots randomly email support@comma.ai

adeebshihadeh, 2020-08-19:
I reviewed more of your routes, and this isn't related to the updater. The ignition line goes low, and your device goes offroad. I recommend unplugging everything and ensuring each connector is seated properly. If the issue persists after that, contact support@comma.ai.

closed 2020-08-19 by adeebshihadeh