commaai/openpilot #2045: Alert that I need to immediately connect to the internet to check for updates, while connected

danhd123, 2020-08-19:
**Describe the bug**

Every time I plug in my comma two, after I stop driving, I get an alert, “Inmediately connect to the internet to check for updates. If you do not connect to the internet, openpilot won’t engage in 0 days”. I am connected to the internet (Sidebar says vehicle online), with 4 dots of cell signal, and via WiFi. It used to be a red warning banner, now it’s blue. 

**How to reproduce or log data**

Happens every time I run OpenPilot. 

**Expected behavior**

It should update?

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f9377828aaadd35e
 - Route: Every route
 - Timestamp: the end of every drive
 - Version: v0.7.7-release
 - Car make/model: 2020 Toyota Corolla

**Additional context**

This is probably a duplicate of one of the later instances of #1004, but the suggestions short of reinstalling openpilot don’t work, and, well, I can’t figure out how to reinstall it from any documentation I’ve found.

adeebshihadeh, 2020-08-19:
Is this the altert still showing? On your last route from today, I’m seeing that it successfully attempted to update today.

adeebshihadeh, 2020-08-20:
Closing due to inactivity and the logs say you’re device updated recently. Feel free to reopen if this happens again or the alert didn’t go away.

closed 2020-08-20 by adeebshihadeh