commaai/openpilot #2090: Following the preceding vehicle

jjg3873, 2020-08-27:
There are cases of unwanted departure from the lane due to following the preceding vehicle. Use the current version 7.8. While driving on the highway, my vehicle followed the preceding vehicle into the rest area while the preceding vehicle entered the rest area.

jjg3873, 2020-08-27:
Unwanted lane departure may occur due to following vehicle. I am currently using version 7.8. While driving on the highway, my vehicle followed the vehicle ahead and entered the rest area.

michaelhonan, 2020-08-28:
@jjg3873 Please edit your issue to use the Bug Template. comma need much more information than what you have provided in order to investigate the issue.

See here: https://github.com/commaai/openpilot/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

closed 2020-09-16 by HaraldSchafer