commaai/openpilot #2093: Frustrated Comma Owner

kyeli2, 2020-08-27:
I got my first Comma2 and it was working well and I wanted it brighter.  I contacted customer service and they said to send it back.  I did and when I received the replacement the EON started unsupported system.  I have written many messages to people in the forum but since I'm new I guess I'm doing it wrong.  I think I have figured out how to go into Workbench and upload but every time I do the screen is either black or it doesn't work.  I new to this and I really need help.  The week that I could use it, it was great.  since then it has been nothing but issues.

JamMarHer, 2020-08-27:
Be patient... Comma is not a product that you can just call and get customer support. They only take care of hardware issues.

VirtuallyChris, 2020-08-28:
Confirmed master-ci fixed this user's issue. They were missing the fpv2 for their car.

closed 2020-08-28 by VirtuallyChris