commaai/openpilot #2100: webcam build error

galenzhao, 2020-08-29:
**Describe the bug**

```
galenzhao@ubuntu:~/openpilot$ scons use_webcam=1
scons: Reading SConscript files ...

scons: warning: Could not detect qt, using moc executable as a hint (QTDIR=/usr)
File "/home/galenzhao/openpilot/SConstruct", line 216, in <module>
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o selfdrive/camerad/main.o -c -std=c++14 -DWEBCAM -g -fPIC -O2 -Wunused -Werror -Wno-deprecated-register -Wno-inconsistent-missing-override -Iexternal/tensorflow/include -I. -Iselfdrive -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal/messaging -Icereal -Iopendbc/can -Iphonelibs/json11 -I/usr/local/include/opencv4 selfdrive/camerad/main.cc
clang++ -o selfdrive/camerad/imgproc/utils.o -c -std=c++14 -DWEBCAM -g -fPIC -O2 -Wunused -Werror -Wno-deprecated-register -Wno-inconsistent-missing-override -Iexternal/tensorflow/include -I. -Iselfdrive -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal/messaging -Icereal -Iopendbc/can -Iphonelibs/json11 -I/usr/local/include/opencv4 selfdrive/camerad/imgproc/utils.cc
clang++ -o selfdrive/camerad/cameras/camera_webcam.o -c -std=c++14 -DWEBCAM -g -fPIC -O2 -Wunused -Werror -Wno-deprecated-register -Wno-inconsistent-missing-override -Iexternal/tensorflow/include -I. -Iselfdrive -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal/messaging -Icereal -Iopendbc/can -Iphonelibs/json11 -I/usr/local/include/opencv4 selfdrive/camerad/cameras/camera_webcam.cc
In file included from selfdrive/camerad/cameras/camera_webcam.cc:13:
In file included from /usr/local/include/opencv4/opencv2/opencv.hpp:52:
In file included from /usr/local/include/opencv4/opencv2/core.hpp:54:
In file included from /usr/local/include/opencv4/opencv2/core/base.hpp:58:
In file included from /usr/local/include/opencv4/opencv2/core/cvstd.hpp:81:
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function
      'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::PaniniPortraitWarper>, void, cv::detail::PaniniPortraitWarper
      *>::check<cv::DefaultDeleter<cv::detail::PaniniPortraitWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function 'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::FisheyeWarper>,
      void, cv::detail::FisheyeWarper *>::check<cv::DefaultDeleter<cv::detail::FisheyeWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function
      'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::CompressedRectilinearWarper>, void, cv::detail::CompressedRectilinearWarper
      *>::check<cv::DefaultDeleter<cv::detail::CompressedRectilinearWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function 'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::MercatorWarper>,
      void, cv::detail::MercatorWarper *>::check<cv::DefaultDeleter<cv::detail::MercatorWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function
      'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::SphericalWarper>, void, cv::detail::SphericalWarper
      *>::check<cv::DefaultDeleter<cv::detail::SphericalWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function
      'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::CompressedRectilinearPortraitWarper>, void, cv::detail::CompressedRectilinearPortraitWarper
      *>::check<cv::DefaultDeleter<cv::detail::CompressedRectilinearPortraitWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
    typedef decltype(check<C>(0)) type;
                     ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:47:60: error: inline function
      'cv::sfinae::has_parenthesis_operator<cv::DefaultDeleter<cv::detail::TransverseMercatorWarper>, void, cv::detail::TransverseMercatorWarper
      *>::check<cv::DefaultDeleter<cv::detail::TransverseMercatorWarper> >' is not defined [-Werror,-Wundefined-inline]
    template<typename> static CV_CONSTEXPR std::false_type check(...);
                                                           ^
/usr/local/include/opencv4/opencv2/core/cvstd_wrapper.hpp:49:22: note: used here
[... 153 lines cut ...]
--     Custom HAL:                  NO
--     Protobuf:                    build (3.5.1)
--
--   OpenCL:                        YES (no extra features)
--     Include path:                /home/galenzhao/opencv-4.3.0/3rdparty/include/opencl/1.2
--     Link libraries:              Dynamic load
--
--   Python 3:
--     Interpreter:                 /home/galenzhao/.pyenv/shims/python3 (ver 3.8.2)
--     Libraries:                   /home/galenzhao/.pyenv/versions/3.8.2/lib (ver 3.8.2)
--     numpy:                       /home/galenzhao/.pyenv/versions/3.8.2/lib/python3.8/site-packages/numpy/core/include (ver 1.19.1)
--     install path:                lib/python3.8/site-packages/cv2/python-3.8
--
--   Python (for build):            /home/galenzhao/.pyenv/shims/python3
--
--   Java:
--     ant:                         NO
--     JNI:                         NO
--     Java wrappers:               NO
--     Java tests:                  NO
--
--   Install to:                    /usr/local
-- -----------------------------------------------------------------
--
```

galenzhao, 2020-08-29:
the content of opencv header file:

```
namespace sfinae {
template<typename C, typename Ret, typename... Args>
struct has_parenthesis_operator
{
private:
    template<typename T>
    static CV_CONSTEXPR std::true_type check(typename std::is_same<typename std::decay<decltype(std::declval<T>().operator()(std::declval<Args>()...))>::type, Ret>::type*);

    template<typename> static CV_CONSTEXPR std::false_type check(...);

    typedef decltype(check<C>(0)) type;

public:
#if __cplusplus >= 201103L || (defined(_MSC_VER) && _MSC_VER >= 1900/*MSVS 2015*/)
    static CV_CONSTEXPR bool value = type::value;
#else
    // support MSVS 2013
    static const int value = type::value;
#endif
};
} // namespace sfinae

```

adeebshihadeh, 2020-10-17:
The webcam build should be fixed and tested in CI as of f4e18e0b7408bc73ffa9320c8ce2689b74993430.

closed 2020-10-17 by adeebshihadeh