commaai/openpilot #2103: 2020 Rav4 Prime gets CAN Error even after the fingerprint is added (STEERING_LKA and PRE_COLLISION seems to have a different checksum scheme?)

nelsonjchen, 2020-08-30:
**Describe the bug**

2020 Rav4 Prime gets CAN Error even [after the fingerprint is added](https://github.com/commaai/openpilot/pull/2018). 

[We've checked the setup against another Rav4 and the harness and C2 do not show the same error message and believe the hardware to be sound.](https://discordapp.com/channels/469524606043160576/524327905937850394/749377789723148368)

We get this error:

[image]

**How to reproduce or log data**

With a known-good Harness + C2 setup installed in the Rav4 Prime, checkout this branch on the C2: https://github.com/nelsonjchen/openpilot/tree/rav4_prime_fw @ commit 640ae8b

Start the Rav4 Prime.

**Expected behavior**

No CAN Error or `CHECKSUM FAIL` error in `tmux`.

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 54bb533aec9ff844
 - Route: 54bb533aec9ff844\|2020-08-29--13-31-50
 - Timestamp: 158433520979974 from logs is when the error message is printed to the screen.
 - Version: https://github.com/nelsonjchen/openpilot/tree/rav4_prime_fw @ 640ae8b0b56a52a4eef662192967546e660c66a0
 - Car make/model: 2020 RAV4 Prime

**Additional context**

I've been helping matty#8553 on Discord add support for his 2020 Rav4 Prime to openpilot. Many others and I thought this was as simple as adding the [fingerprint for the TSS2 Rav4 Prime to OP](https://github.com/commaai/openpilot/pull/2018). This turns out to not be the case. 

After reproduction, we have SSH'd into the C2 and taken a look at `tmux a`, we see `CHECKSUM FAIL` spam:

[image]

Looking at Cabana, it seems that `STEERING_LKA` is now 8 bytes in size up from 5 bytes in the past and that PRE_COLLISION IS size 8 up from 7.

I took a look at `STEERING_LKA` and tried the existing checksum algorithm [in a scratch-program, I implemented Toyota's checksum per the IOActive document to the last value and seeing if that pre-existing approach works for the RAV4 Prime's 8 byte `STEERING_LKA`. My unit tests work for my Corolla 2020's data. It doesn't for the 2020 Rav4 Prime's data and they don't match.](https://github.com/nelsonjchen/toyota_checksum_2020_scratch/blob/8422bd3b4b7770391e940d31202b8129fdebcb02/src/lib.rs#L97-L108)

The last four bytes for `STEERING_LKA` are rapidly changing. 

[image]

I haven't taken a look at `PRE_COLLISION` other than noticing the size change. Using the existing algorithm on a value here doesn't seem to work either.

Unfortunately, I've hit the limits of what I can do right now. I think Toyota has changed the checksum scheme or something for `STEERING_LKA` and `PRE_COLLISION`. 

I am not the owner of this 2020 Rav4 Prime, matty#8553 on Discord is. 

Please reach out to matty#8553 if you need him to perform some diagnosis. If there may be something out of his depth and Comma staff or someone from the community is not available or willing to help him through the "boring" diagnosis parts such as setup/SSH/forks/etc, let me know and I will be happy to help walk matty#8553 through those tasks.

adeebshihadeh, 2020-08-30:
This isn’t an issue - the RAV4 Prime isn’t listed as a supported car.

closed 2020-08-30 by adeebshihadeh

jassdsu, 2020-08-31:
I have the same issue on my 2020 rav4 hybrid- I emailed support. Not sure if there is a solution or even if this is a supported vehicle anymore.

wmoralesap, 2020-09-21:
Did you have any answer? I have the same issue with my 2020 rav4 hybrid.

jassdsu, 2020-09-21:
I have 2020 Hybrid- you need change the program to the master-ci version. You need help with that?

wmoralesap, 2020-09-21:
Yes please! How do I change to the master-ci version?

jassdsu, 2020-09-21:
Firstly, have you verified that someone is using the Prime with Comma AI?  I did have the same error message so it sounds similar.  My parents are awaiting on their Prime to come in so I’ll be figuring this out soon- hopefully.

Start by downloading workbench (you’ll find the link for it on commas website) on your PC, and plugging your Comma unit in somewhere nearby on the same wifi as your PC.  Workbench communicates with the unit and it’ll allow you to update the operating system.

I have to track down the step by step instructions- its really not bad but it took me a while to understand how everything works.

wmoralesap, 2020-09-21:
I already changed to master-ci version but now says dashcam mode: Car unrecognized. 
[image]

[image]

jassdsu, 2020-09-21:
Nice work, now it needs to pair with your vehicle.  Leave the comma on for ten minutes and it should locate it.  If not, restart the comma and give it another 10 minutes.  You can turn your car off while doing this.

wmoralesap, 2020-09-21:
I already tried it. The weirdest thing is that when thr engine is off, the comma 2 device shows vehicle connected. but when I turn it on, goes to dashcam mode. Do I need to fingerprint it?

jassdsu, 2020-09-21:
Did it take about 30 minutes to install master-CI?  It takes a while and I’m wondering if it didn’t finish installing. I had a similar problem before I installed master-ci

wmoralesap, 2020-09-21:
Yeah, installing master-ci took like 30 minutes.

jassdsu, 2020-09-21:
Search for other people with your exact car- I know there’s a few users. Sorry- mine worked after I did all that.

If you find a solution let me know cause I’ll be dealing with this in a month.

wmoralesap, 2020-09-21:
I understand. Thanks for your support! I will reply back if I found a solution

jassdsu, 2020-09-23:
Did you figure anything out?

wmoralesap, 2020-09-23:
Yes, it was the fingerprint for the ESP. They already uploaded the new FW value and it works really good now.

jassdsu, 2020-09-23:
What did you need to do differently? I’m not sure I understand.

jassdsu, 2020-09-23:
> I just download the master-ci again. They uploaded a new version that fixed my problem.

Awesome. Enjoy.

nelsonjchen, 2020-09-23:
@jassdsu @wmoralesap I hate to bother you two but could we keep the conversation here on the 2021 Rav4 Prime and/or the new `STEERING_LKA` message that is currently an unknown? It might be better if that were discussed on Discord (or perhaps if comma (read @VirtuallyChris) were to [ask for and setup GitHub Discussions on this repo](https://github.community/t/can-one-apply-for-github-discussions-beta/121120/226))?

jassdsu, 2020-09-23:
I don’t have the prime, I have the 2020 RAV4 hybrid and I have not experienced that issue. I know there are users with that vehicle that are running master CI and it’s working fine. I would recommend reaching out to them.

Are you currently experiencing the issue or are you just wondering about it?

nelsonjchen, 2020-09-23:
@jassdsu This GitHub issue is about the 2021 Rav4 Prime.

nelsonjchen, 2021-01-27:
I've since started a sheet to organize and record a community bounty for the ECU Security Key issue that is likely the nature of this new "code":

https://docs.google.com/spreadsheets/d/1MKS78_utvbAe74Xv7zszgEnn6JrtBgpgYlVOfoIvLEw/edit#gid=0

jonvkw, 2021-01-31:
Does anyone know if the source of the key is from the front facing camera or ECU?  I was wondering if swapping the front facing camera from another lower year Rav4 may revert to 5 byte LKA message?

adeebshihadeh, 2021-01-31:
Locked the conversation since this issue was closed. Let's move the discussion over to our GitHub Discussions page.

adeebshihadeh, 2021-08-04:
Relevant discussion: https://github.com/commaai/openpilot/discussions/19932