commaai/openpilot #2105: Comma Two Fails to Update

musicaljoeker, 2020-08-30:
**Describe the bug**

My comma two says I need to connect to the internet immediately to check for an update when it's already connected to the internet (wifi from my apartment).

**How to reproduce or log data**

I'm not sure where logs are written, but I can reproduce this by manually modifying the date in the `/data/params/d/LastUpdateTime` file. I can also see a count of 85 failed updates in the `/data/params/d/UpdateFailedCount`.

**Expected behavior**

The device checks for updates and updates to 0.7.8

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 9ca46bc261b779e2
 - Route: Not route specific
 - Timestamp: On boot
 - Version: 0.7.7-release2
 - Car make/model: 2018 Toyota Camry

**Additional context**

I've looked at other issues (#2045, #2016) but none of these had any tangible resolution. I have Linux experience and am willing to help you get to the bottom of this issue as it seems to be impacting several people.

For now, my workaround is to manually modify the LastUpdateTime file to continue to use OP.

adeebshihadeh, 2020-08-30:
The update fails on every attempt because of an orphaned git lock. Remove the lock with `rm -f /data/openpilot/.git/index.lock`, and trigger an update attempt with `cd /data/openpilot && scripts/update_now.sh`.  This is fixed in 0.7.8.

closed 2020-08-30 by adeebshihadeh

musicaljoeker, 2020-08-31:
Thank you for the very quick response (on a Sunday no less). I was able to update the comma two after following your instructions.