commaai/openpilot #2110: OP LKAS fault occurs when LKAS disengages.

alekseymya, 2020-08-31:
**Describe the bug**
When vehicle speed goes below the LKAS threshold, then OP loses its connection with LKAS. The following error occurs: "LKAS fault: Restart car to engage".
Vehicle: Chrysler Pacifica 2020
LKAS threshold: 38 m/h

**How to reproduce or log data**

1. Power up OpenPilot (OP)
2. Drive faster than 38m/h  to engage LKAS 
3. Slow back down to under 38m/h 
4. LKAS error "LKAS fault: Restart car to engage" will appear on the OP screen.
5. The car will indicate that the LKAS system is unavailable.
6. OP functionality is unavailable.

**Expected behavior**
OP should be able to engage and re-engage when the vehicle speeds up to LKAS managed speed range.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 3d84727705fecd04
 - Route: 3d84727705fecd04|2020-08-30--19-27-54
 - Timestamp: ~21:30
 - Version: 
openpilot v0.7.8-release 
Git branch: release2
Git revision: e164f7f99576
Panda Firmware: bdcd195d8521b9d

 - Car make/model: Chrysler Pacifica 2020

**Additional context**
When attempted to manually turn LKAS back on (button) after this error happens then LKAS turns off automatically.
Sometimes the car restart doesn't solve the problem. Panda dongle also has to be pulled out and re-inserted. 

See attached Screenshots. 
[image]
[image]

adhintz, 2020-08-31:
What car model is OpenPilot fingerprinting your car as?

My guess is that you're being fingerprinted as a Pacifica 2017 or 2018. I'd like a feature that shows in the OpenPilot UI what car you're fingerprinted as.

To tell, ssh in to the Comma2 before you turn on your car and run `tmux a`  Then as you turn on your car watch the scrolling text (and page back up as needed) to see what your car fingerprints as.

You'll probably need to run these commands to generate a fingerprint: https://github.com/commaai/openpilot/wiki/Fingerprinting#fingerprinting-10
Feel free to paste your fingerprint here and I'll see how it lines up with other models.

pd0wm, 2020-08-31:
The posted route has ` carFingerprint = "CHRYSLER PACIFICA 2020",`

alekseymya, 2020-08-31:
Thank you @pd0wm  
( key = "LiveParameters",
          value = "{\"carFingerprint\": \"CHRYSLER PACIFICA 2020\", \"steerRatio\": 11.61391544342041, \"stiffnessFactor\": 0.9674015045166016, \"angleOffsetAverage\": -1.4258414506912231}" ),

@adhintz
I will also follow your instructions and see if I can get more useful data.

adhintz, 2020-08-31:
Thanks for getting the fingerprint data from the route! Sounds like it's not a fingerprint issue.

If you'd like, feel free to get the Cabana link, click Copy Share Link, and send the link to me or post it here. That will let anyone with the link view your drive. If you'd like to look through it yourself, a few things I'd look for around the time of the LKAS fault:
- When you go below 38mph, that the LKAS_COMMAND 0x292 (658) messages sent by OpenPilot (on the 128+ bus) go down to 0 torque.
- That there are no gaps in the LKAS messages sent by OpenPilot, even a gap of 250ms will cause an LKAS fault.
- To find the time of the LKAS fault, look for when EPS_STATUS in LKAS_STATE goes to 4 or down to 0 while still going fast.

VirtuallyChris, 2020-08-31:
I've had a few users reach out about this same bug, I recommended they add to this issue.

alekseymya, 2020-09-04:
Run a better test at 17:15:04 https://my.comma.ai/3d84727705fecd04/1599177756130/1599179548099#

DubiousLatchkey, 2020-10-03:
I've also had this problem on a 2018 Pacifica as well when pressing the pedal while openpilot is engaged and at a full stop.

vanillagorillaa, 2021-01-10:
Is this issue still relevant? The data from the provided drives has been deleted. If anyone still has this issue please upload a route and be sure to preserve it in useradmin (my.comma.ai/useradmin) once fully uploaded.

qrld23, 2021-01-19:
b66bf349eecf2fda|2021-01-19--17-21-14--7

from my experience on my 2021 PACIFICA HYBRID, there are 3 ways to create this issue:
1. have ACC (and OP) enabled and set at any speed below 39(?)
2. press accelerator with ACC and OP enabled and set
3. press brake with ACC and OP enabled and set

note: i do not get the "LKAS fault" on my comma2. i just get the "service ACC" warning msg and light on my dash. must turn off and turn on vehicle to reset. 
PS i am "user2323" on discord

niewiadomcio, 2021-01-20:
I have a 2020 Hybrid Limited S bought in December of 2020 and get the same error. I don’t get it every time the car crosses 39mph, while slowing down. I believe it happens when the car’s ACC is braking hard through the 39mph threshold. I have a ride when this happened most recently preserved. I do not get an LKAS fault on my comma2 screen. I get one or two messages on the car display: “FCW active braking off” and “ACC/FCW Unavailable Service Required.” c424280107718f16|2021-01-17--13-38-51 time stamp: 14:10:45

niewiadomcio, 2021-01-20:
[image]
[image]

qrld23, 2021-01-20:
same exact thing as @niewiadomcio. seems like its probably the same issue as described in bug details, but some cars dont throw the error to the OP.

AlligatorDreams, 2021-01-20:
This is an issue with older/non Hybrid Pacificas as well. I have a 2017 Gas Pacifica and experienced this same fault yesterday. Occurred after engaging OP while at a full stop with a vehicle in front of me. I will look into getting my drive data uploaded and will update.

alekseymya, 2021-01-21:
@vanillagorillaa The error still happens here is the most recent route:
https://my.comma.ai/3d84727705fecd04/1611198034179/1611198451410
https://my.comma.ai/useradmin/?onebox=3d84727705fecd04%7C2021-01-20--19-00-34

adhintz, 2021-03-07:
I figured out at least one LKAS fault condition. Have the car stopped with OP engaged, such as behind a stopped car. Then press the accel pedal and cross the minSteerSpeed value. In Cabana it appears that the lkas commands are then not sent for half a second, which will cause an LKAS fault. This points to Panda's safety code preventing the transmit.

Panda `generic_rx_checks` sets `controls_allowed  = 0` on `gas_pressed`. Maybe it's new, or I didn't realize it at the time, but these generic checks are called for each car type, including chrysler.

The gas_resume_speed value comes from chrysler/interface.py and is set to 2. In the Panda safety_chrslyer.h it's set to
`const int CHRYSLER_GAS_THRSLD = 30; // 7% more than 2m/s`
The goal of the 7% appears to be to so that OP has time to cancel the ACC before Panda forces it.

However, it doesn't work in this situation due to logic in car/interfaces.py for sending pedalPressed to cause a disable:
`if (cs_out.gasPressed and (not self.CS.out.gasPressed) and cs_out.vEgo > gas_resume_speed)`

When it's first accelerating from zero, the gas_resume_speed speed check fails, so it doesn't disable ACC. However cs_out.gasPressed is then set to true, and so on the next and all future iterations, self.CS.out.gasPressed will be true. My understanding is that self.CS.out.gasPressed is the CarState from the previous interface update() call. Therefore the ACC cancel is never sent by OP. However Panda expects it to be canceled once it reaches  CHRYSLER_GAS_THRSLD, so it starts blocking messages.

 This issue only occurs on Chrysler because it's the only make that sets gas_resume_speed to something other than the -1 default.

To fix this, two ideas:
1) In car/interfaces.py if it's going below gas_resume_speed, then set the cs_out.gasPressed back down to false. This way it will cancel ACC as soon as it crosses gas_resume_speed.
2) Implement Resume spamming when the car is stopped. This would change the behavior of OP to be different than FCA stock ACC, but the same as other makes on OP. This comment implies it's what Comma wants:
`# e.g. Chrysler does not spam the resume button yet, so resuming with gas is handy. FIXME!`
This would allow OP to remove the gas_resume_speed code and the special-casing of Chrysler. However it would need the Chrysler Resume spamming to work perfectly -- to do that it would probably need to track and use the frame value of 23b.

Any preferences for how to fix it? If resume spamming is the preferred way, are there any existing makes that do resume spamming?

j-vanetten, 2021-03-07:
> 2. Implement Resume spamming when the car is stopped. This would change the behavior of OP to be different than FCA stock ACC, but the same as other makes on OP. This comment implies it's what Comma [...]
Funny.  I was just looking at this code yesterday to see if I could get this working by spamming the Resume button when the brake is let go.

imadethem1, 2021-03-08:
I'd vote for the resume spamming option. Looks like they use it for Hyundais, or used to. They may have found a better solution that prevents the auto resume from timing out

chandru-jsr, 2021-04-04:
> from my experience on my 2021 PACIFICA HYBRID, there are 3 ways to create this issue:
> 
> 1. have ACC (and OP) enabled and set at any speed below 39(?)
> 2. press accelerator with ACC and OP enabled and set
> 3. press brake with ACC and OP enabled and set (only sometimes will this cause the error)
> 
> note: i do not get the "LKAS fault" on my comma2. i just get the "service ACC" warning msg and light on my dash. must turn off and turn on vehicle to reset.
> PS i am "user2323" on discord
> 
> here is an example of using accel pedal:
> b66bf349eecf2fda|2021-01-19--17-21-14--7

Thanks for posting the message. In short, I see that everytime you mistakenly press brake below 39, it ask you to restart the car. 

1. Is overall OP in Pacifica 2021 has nice experience?
2. Can you continuously able to drive in OP for long time or is it annoying with this error everytime?

alekseymya, 2021-04-06:
I have verified that the LKAS fault issue with 2020 Pacifica is fixed. Thank you @adhintz, you rock.

vanillagorillaa, 2021-04-06:
Anyone still having this issue with latest release? If not I will close.

ZeroAlligator, 2021-04-06:
I am in the process of testing the various scenarios that caused this in my 2017 Pacifica. I’ll update my findings later tonight.

ZeroAlligator, 2021-04-07:
I confirmed that this bug is still an issue with my 2017. I can recreate it the vehicle comes to a complete stop in traffic with Open Pilot engaged. When traffic starts moving, Open Pilot fails to accelerate from a stop, and an LKAS fault occurs upon pressing the gas pedal in order to move the vehicle forward. Here is a short drive where this occurs: https://my.comma.ai/410c721d6d35265f

chandru-jsr, 2021-04-10:
Hi, Anyone using Pacifica 2021 with open pilot? I am planning to buy one only if it is useful with open pilot, else it is waste with ping pong. Please suggest.

drneet, 2021-04-19:
I still have this issue on pacifica hybrid 2018. Updated OP to latest version few days ago.

alekseymya, 2021-04-21:
I think that it makes sense to file a new similar issue specifically for Pacifica 2017-2018.

ZeroAlligator, 2021-04-21:
Agreed, it does indeed sound like this issue with the Chrysler Pacifica/Pacifica Hybrid 2020+ has been resolved and this issue can be closed. I will add a new issue for the 2017-2018 Chrysler Pacifica LKAS Faults.

ZeroAlligator, 2021-04-21:
Added additional LKAS issue for 2017+ as: https://github.com/commaai/openpilot/issues/20724

closed 2021-04-21 by alekseymya

werkhovl, 2021-05-08:
> Anyone still having this issue with latest release? If not I will close.

Yes, just installed OP in 2020 Pac Hyb. And has LKAS error mentioned in this thread in all these mentioned scenarios.

werkhovl, 2021-05-08:
> Hi, Anyone using Pacifica hybrid 2021 with open pilot? I am planning to buy one only if it is useful with open pilot, else it is waste with ping pong. Please suggest.
> @qrld23 - please help

I installed yesterday on 2020 pacifica hybrid and haven't experienced any improvement over stock LKAS.  Not sure what is wrong.  Just an expensive giant dashcam at this point.   I haven't found any basic troubleshooting advise anywhere as these threads seem to be for programmers.  I may return if I can't figure out.

j-vanetten, 2021-05-08:
> I installed yesterday on 2020 pacifica hybrid and haven't experienced any improvement over stock LKAS. Not sure what is wrong. Just an expensive giant dashcam at this point. I haven't found any [...]

@werkhovl Check out the Chrysler community channel on discord and we all can help troubleshoot:  https://discord.com/channels/469524606043160576/525364884410728462

werkhovl, 2021-05-08:
> Hi, Anyone using Pacifica hybrid 2021 with open pilot? I am planning to buy one only if it is useful with open pilot, else it is waste with ping pong. Please suggest.
> @qrld23 - please help

It works well.  Had to uninstall then install the advanced rather than default software

werkhovl, 2021-05-08:
Thanks, looks like I needed to install the advanced rather than default
software. Working great.

werkhovl, 2021-05-09:
Thanks,  I uninstalled the defsult dashcam software on the EON dev and installed the advanced OpenPilot and its working great now

ns3284, 2022-04-21:
Installed in 2022 Pacifica Hybrid, and when ACC brings the speed below 39mph, I am getting a faulted ACC/LKAS system, that requires a reboot.

Hardware Comma 3, Release3: 0.8.13

sshane, 2022-07-01:
@ns3284 are you still experiencing this, what is your dongle id? You must be on a fork, as 2022 support isn't in upstream openpilot yet.

ns3284, 2022-07-01:
@sshane So far, so good. On master-ci branch.

Fritzie5, 2022-12-03:
Hello i just installed and am getting the lkas fault code on my 2017 Pacifica tried the discord page which didn’t work. Please can someone in here help.

ZeroAlligator, 2022-12-03:
Yes, I still have LKAS faults on my 2017 Pacifica as well, but only if I don’t disengage when the vehicle is stopped. My workaround is to disengage when stopping/stopped. Works fine otherwise 😁

Fritzie5, 2022-12-03:
But how do you do that my lkas generates right away

Fritzie5, 2022-12-03:
Than it shuts off the stock lane keep assist and acc

ZeroAlligator, 2022-12-03:
Huh, I’m not sure. I’m using the main branch and that doesn’t happen to me. Do you still have a fault when you turn the van completely off, open and close the door, wait 30 seconds (for the dashboard lights to turn off), then start the van again? That’s the LKAS clearing sequence that I do when it occurs.

Fritzie5, 2022-12-03:
Yeah perhaps it’s a hardware thing. I just need someone that writes the code to ssh in and fix things hah