commaai/openpilot #2120: OpenGL on UI view not working

galenzhao, 2020-09-02:
**Describe the bug**

OpenGL on UI view not working.
black view with lots of error logs.
```
Error 00000502 after convex fill
``` 

**How to reproduce or log data**
build follow webcam guide.
start with:
```
PASSIVE=0 NOSENSOR=1 WEBCAM=1 ./manager.py
```

and the error log [Error 00000502 after convex fill] repeat forever.
```
galenzhao@ubuntu:~/openpilot/selfdrive$ PASSIVE=0 NOSENSOR=1 WEBCAM=1 ./manager.py 
getting pilotauth
dongle id is 9d514655c5dec38d
dirty is 1
preimporting selfdrive.thermald.thermald
Building panda firmware
preimporting selfdrive.loggerd.uploader
preimporting selfdrive.loggerd.deleter
preimporting selfdrive.controls.controlsd
preimporting selfdrive.controls.plannerd
preimporting selfdrive.controls.radard
preimporting selfdrive.monitoring.dmonitoringd
preimporting selfdrive.logmessaged
preimporting selfdrive.locationd.locationd
preimporting selfdrive.tombstoned
preimporting selfdrive.pandad
preimporting selfdrive.locationd.calibrationd
preimporting selfdrive.locationd.paramsd
preimporting selfdrive.updated
manager start
{'environ': environ({'XDG_VTNR': '2', 'XDG_SESSION_ID': '1', 'SSH_AGENT_PID': '1588', 'CLUTTER_IM_MODULE': 'xim', 'PYENV_ROOT': '/home/galenzhao/.pyenv', 'VTE_VERSION': '4205', 'XDG_MENU_PREFIX': 'gnome-', 'SHELL': '/bin/bash', 'TERM': 'xterm-256color', 'WEBCAM': '1', 'QT_LINUX_ACCESSIBILITY_ALWAYS_ON': '1', 'GJS_DEBUG_OUTPUT': 'stderr', 'WINDOWID': '27262982', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'GTK_MODULES': 'gail:atk-bridge', 'PYENV_VERSION': '3.8.2', 'USER': 'galenzhao', 'QT_ACCESSIBILITY': '1', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'SSH_AUTH_SOCK': '/tmp/ssh-yojVkMlRu6SS/agent.1520', 'PYENV_DIR': '/home/galenzhao/openpilot/selfdrive', 'DEFAULTS_PATH': '/usr/share/gconf/gnome.default.path', 'USERNAME': 'galenzhao', 'SESSION_MANAGER': 'local/ubuntu:@/tmp/.ICE-unix/1520,unix/ubuntu:/tmp/.ICE-unix/1520', 'PYENV_VIRTUALENV_INIT': '1', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-gnome:/etc/xdg', 'NOSENSOR': '1', 'DESKTOP_SESSION': 'gnome', 'PATH': '/home/galenzhao/.pyenv/versions/3.8.2/bin:/home/galenzhao/.pyenv/libexec:/home/galenzhao/.pyenv/plugins/python-build/bin:/home/galenzhao/.pyenv/plugins/pyenv-virtualenv/bin:/home/galenzhao/.pyenv/plugins/pyenv-update/bin:/home/galenzhao/.pyenv/plugins/pyenv-installer/bin:/home/galenzhao/.pyenv/plugins/pyenv-doctor/bin:/home/galenzhao/.pyenv/shims:/home/galenzhao/.pyenv/plugins/pyenv-virtualenv/shims:/home/galenzhao/.pyenv/bin:/home/galenzhao/.pyenv/plugins/pyenv-virtualenv/shims:/home/galenzhao/.pyenv/shims:/home/galenzhao/.pyenv/bin:/home/galenzhao/bin:/home/galenzhao/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'QT_IM_MODULE': 'ibus', 'PWD': '/home/galenzhao/openpilot/selfdrive', 'XDG_SESSION_TYPE': 'x11', 'XMODIFIERS': '@im=ibus', 'LANG': 'en_US.UTF-8', 'MANDATORY_PATH': '/usr/share/gconf/gnome.mandatory.path', 'PYENV_HOOK_PATH': '/home/galenzhao/.pyenv/pyenv.d:/usr/local/etc/pyenv.d:/etc/pyenv.d:/usr/lib/pyenv/hooks:/home/galenzhao/.pyenv/plugins/pyenv-virtualenv/etc/pyenv.d:/home/galenzhao/.pyenv/plugins/pyenv-which-ext/etc/pyenv.d', 'GDMSESSION': 'gnome', 'SHLVL': '1', 'HOME': '/home/galenzhao', 'XDG_SEAT': 'seat0', 'PYENV_SHELL': 'bash', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'LOGNAME': 'galenzhao', 'XDG_SESSION_DESKTOP': 'gnome', 'PYTHONPATH': '/home/galenzhao/openpilot', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-7LmXk2EBy0,guid=b463a7f1e0a030273008c5ad5f4f0151', 'QT4_IM_MODULE': 'xim', 'XDG_DATA_DIRS': '/usr/share/gnome:/usr/local/share:/usr/share:/var/lib/snapd/desktop', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'PASSIVE': '0', 'WINDOWPATH': '2', 'DISPLAY': ':1', 'XDG_RUNTIME_DIR': '/run/user/1000', 'OPENPILOT_ENV': '1', 'GTK_IM_MODULE': 'ibus', 'XDG_CURRENT_DESKTOP': 'GNOME', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'BASEDIR': '/home/galenzhao/openpilot', 'PARAMS_PATH': '/home/galenzhao/openpilot/persist/params', 'DONGLE_ID': '9d514655c5dec38d'})}
selfdrive/loggerd/loggerd.cc: bootlog to /data/media/0/realdata/2020-09-01--23-46-16--0
starting python selfdrive.thermald.thermald
starting python selfdrive.logmessaged
starting process ui
starting python selfdrive.loggerd.uploader
starting python selfdrive.pandad
uploader_fn
thermal: Reader was evicted, reconnecting
Connecting to panda
resize 1920x1080
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
sh: 1: cannot create /sys/class/devfreq/soc:qcom,cpubw/governor: Directory nonexistent
{"event": "STATUS_PACKET", "count": 0, "health": null, "location": null, "thermal": {"thermal": {"cpu0DEPRECATED": 0, "cpu1DEPRECATED": 0, "cpu2DEPRECATED": 0, "cpu3DEPRECATED": 0, "memDEPRECATED": 0, "gpuDEPRECATED": 0, "batDEPRECATED": 0, "freeSpace": 0.7387802004814148, "batteryPercent": 100, "batteryStatus": "Charging", "fanSpeed": 0, "started": false, "usbOnline": false, "startedTs": 0, "thermalStatus": "green", "batteryCurrent": 0, "batteryVoltage": 0, "chargingError": false, "chargingDisabled": false, "memUsedPercent": 23, "cpuPerc": 30, "pa0DEPRECATED": 0, "networkType": "wifi", "offroadPowerUsage": 0, "networkStrength": "unknown", "carBatteryCapacity": 3000000, "cpu": [0.0], "gpu": [0.0], "mem": 0.0, "bat": 0.0, "ambient": 0.0}, "logMonoTime": 5237258866695, "valid": true}}
thermald logmessaged ui uploader pandad
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill
```

**Expected behavior**

road camera image shows on UI view.

**Additional context**

Operating system: [e.g. Ubuntu 16.04]

pd0wm, 2020-09-02:
I just added a print statement on ui startup that prints the opengl version (https://github.com/commaai/openpilot/commit/92de562ed91594d20f10d3b9a30d499c801baeff). What does that print for you?

What hardware/gpu are you using? Nvidia/amd or intel integrated graphics?

adeebshihadeh, 2020-09-13:
Closing due to inactivity

galenzhao, 2020-09-16:
```
galenzhao@ubuntu:~/openpilot/selfdrive/ui$ ./ui
OpenGL version: OpenGL ES 3.2 Mesa 18.0.5
OpenGL vendor: Intel Open Source Technology Center
OpenGL renderer: Mesa DRI Intel(R) HD Graphics 615 (Kaby Lake GT2) 
OpenGL language version: OpenGL ES GLSL ES 3.20
resize 1920x1080
Error 00000502 after convex fill
Error 00000502 after convex fill
Error 00000502 after convex fill

```

> I just added a print statement on ui startup that prints the opengl version ([92de562](https://github.com/commaai/openpilot/commit/92de562ed91594d20f10d3b9a30d499c801baeff)). What does that print for you?
> 
> What hardware/gpu are you using? Nvidia/amd or intel integrated graphics?

sohamm17, 2020-10-10:
Happens to me as well, when I was trying to run the simulated version. I have no Nvidia GPU on my laptop, only Intel integrated graphics. I tried to run `PASSIVE="0" NOBOARD="1" SIMULATION="1" ./manager.py`

This is the output:

```
OpenGL version: OpenGL ES 3.2 Mesa 18.0.5
OpenGL vendor: Intel Open Source Technology Center
OpenGL renderer: Mesa DRI Intel(R) Iris Pro Graphics 580 (Skylake GT4e) 
OpenGL language version: OpenGL ES GLSL ES 3.20
resize 1920x1080
resize 1855x1056
resize 1920x1080
Error 00000502 after convex fill
Error 00000502 after convex fill
```

pd0wm, 2020-10-12:
Does this happen immediately on startup of the ui, or only if it's trying to draw the camera view?

sohamm17, 2020-10-12:
> Does this happen immediately on startup of the ui, or only if it's trying to draw the camera view?
> 
> @sohamm17 Does your PR (#2315) fix it?

The error appears as soon as the UI starts running. The problem happens at [this line](https://github.com/commaai/openpilot/blob/0365db20d8853845dce79727e347b02ec07b472c/selfdrive/ui/paint.cc#L630). I am not sure the reason behind the error. My [PR](https://github.com/commaai/openpilot/pull/2315) does not fix this error.

The UI was crashing when it was connected to the Carla via the bridge. My PR fixes that issue. The above error still comes, just after the start and even when the vision system is correctly connected to Carla.

adeebshihadeh, 2021-02-26:
Is this still an issue?

closed 2021-02-28 by adeebshihadeh