commaai/openpilot #2127: Dashcam Car Unrecognized - 2018 Grand Cherokee

atstevenha, 2020-09-03:
Dash cam mode car Unrecognized error. 

<!-

**How to reproduce or log data**

**Expected behavior**

Clear dashcam and able to use open pilot,  before it worked without no connection to the OBD 2. The new update doesn't work for me. When would the 2018 Grand Cherokee able to connect to the OBD 2? Thanks in advance 

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: 8551ebbdd971b5c8
 
 - Timestamp: When I start the car
 - Version: 0.7.8
 - Car make/model: Jeep Grand Cherokee TrailHawk 2018

**Additional context**

atstevenha, 2020-09-05:
Please advice. Thank you

atstevenha, 2020-09-07:
Is something wrong with my post?

atstevenha, 2020-09-11:
I don't understand what's wrong. I've tried everything and read the entire forum on this issue. This stupid system isn't working since the update.

cydia2020, 2020-09-11:
is this a duplication of #1996 ?

Did you change branch?

atstevenha, 2020-09-11:
> is this a duplication of #1996 ?
> 
> Did you change branch?

Thanks for replying.  I haven't changed anything. The cable male is a bit bent but don't think that's it.

cydia2020, 2020-09-12:
Hi,

In #1996 in the photo you've taken the branch on it clearly showed chffrplus.

https://github.com/commaai/openpilot/issues/1996#issuecomment-670800160

Chffrplus is openpilot's dashcam software and doesn't control the vehicle.

You need to uninstall chffrplus from I think settings - device, someone correct me if I'm wrong here, and during the setup select <b>custom software</b> and put openpilot.comma.ai in the address bar

atstevenha, 2020-09-12:
> Hi,
> 
> In #1996 in the photo you've taken the branch on it clearly showed chffrplus.
> 
> [#1996 (comment)](https://github.com/commaai/openpilot/issues/1996#issuecomment-670800160)
> 
> Chffrplus is openpilot's dashcam software and doesn't control the vehicle.
> 
> You need to uninstall chffrplus from I think settings - device, someone correct me if I'm wrong here, and during the setup select custom software and put https://openpilot.comma.ai in the address bar
> 
> Edit: the aforementioned steps might not solve your problem, but let's start from here.

That is strange. I'll try it again but I swear I only use https://openpilot.comma.ai/. Thanks!

atstevenha, 2020-09-15:
I'm starting to think its my cable. Its bent on accident. 
[image]

cydia2020, 2020-09-19:
Yeah looks like it, get a new one

adeebshihadeh, 2020-09-22:
There was a bug (fixed in a81a18205e85ac576a5cf5e330e3f7b99fcb4152) where community features cars would show the dashcam startup alert if the toggle wasn't enabled. Does your car still fault if you unplug the OBD C cable while leaving the harness intact?

atstevenha, 2020-09-22:
Thank you everyone for replying back. Chris @ Comma.ai has sent me a new cable, I have tried every cable but it doesn't show the cam so I figured it must be a special cable that isn't just data. Once I receive the cable, Ill update if there are any issues if it works. Chris mention that the 2018 Grand Cherokee can now run the OBD2 without being locked from the security relay. Ill try to have everything plugged in. Do I need to do anything special in the software to get the OBD2 to work? Thanks again, I can't wait to have this working again. =)

closed 2020-09-22 by adeebshihadeh

visionandy, 2022-05-08:
Is this problem still on at your vehicle? I always have this problem, sometimes it's more frequent, sometime not. Recently, it appears more often.