commaai/openpilot #2128: Docker Ubuntu Build Script Failing

giithob, 2020-09-04:
**Describe the bug**

Docker build script fails.

**How to reproduce or log data**

```
user2@user-All-Series:~/openpilot/tools$ bash ubuntu_setup.sh 
[sudo] password for user2: 
Hit:1 http://us.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease              
Get:3 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease [98.3 kB]  
Hit:4 https://download.docker.com/linux/ubuntu focal InRelease                 
Get:5 http://security.ubuntu.com/ubuntu focal-security InRelease [107 kB]      
Fetched 205 kB in 0s (416 kB/s)    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
E: Unable to locate package python-pip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   148  100   148    0     0   1042      0 --:--:-- --:--:-- --:--:--  1042
100  2454  100  2454    0     0  12331      0 --:--:-- --:--:-- --:--:-- 12331
Cloning into '/home/user2/.pyenv'...
remote: Enumerating objects: 707, done.
remote: Counting objects: 100% (707/707), done.
remote: Compressing objects: 100% (545/545), done.
remote: Total 707 (delta 366), reused 246 (delta 70), pack-reused 0
Receiving objects: 100% (707/707), 392.16 KiB | 2.29 MiB/s, done.
Resolving deltas: 100% (366/366), done.
Cloning into '/home/user2/.pyenv/plugins/pyenv-doctor'...
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 11 (delta 1), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (11/11), 38.62 KiB | 1.84 MiB/s, done.
Cloning into '/home/user2/.pyenv/plugins/pyenv-installer'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 16 (delta 1), reused 8 (delta 0), pack-reused 0
Unpacking objects: 100% (16/16), 5.99 KiB | 876.00 KiB/s, done.
Cloning into '/home/user2/.pyenv/plugins/pyenv-update'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 10 (delta 1), reused 6 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), 2.66 KiB | 1.33 MiB/s, done.
Cloning into '/home/user2/.pyenv/plugins/pyenv-virtualenv'...
remote: Enumerating objects: 57, done.
remote: Counting objects: 100% (57/57), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 57 (delta 11), reused 20 (delta 0), pack-reused 0
Unpacking objects: 100% (57/57), 34.86 KiB | 2.90 MiB/s, done.
Cloning into '/home/user2/.pyenv/plugins/pyenv-which-ext'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 10 (delta 1), reused 6 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), 2.92 KiB | 995.00 KiB/s, done.

WARNING: seems you still have not added 'pyenv' to the load path.

# Load pyenv automatically by adding
# the following to ~/.bashrc:

export PATH="/home/user2/.pyenv/bin:$PATH"
eval "$(pyenv init -)"
eval "$(pyenv virtualenv-init -)"
added openpilot_env to bashrc
Submodule 'cereal' (https://github.com/commaai/cereal.git) registered for path 'cereal'
Submodule 'laika_repo' (https://github.com/commaai/laika.git) registered for path 'laika_repo'
Submodule 'opendbc' (https://github.com/commaai/opendbc.git) registered for path 'opendbc'
Submodule 'panda' (https://github.com/commaai/panda.git) registered for path 'panda'
Submodule 'rednose_repo' (https://github.com/commaai/rednose.git) registered for path 'rednose_repo'
Cloning into '/home/user2/openpilot/cereal'...
Cloning into '/home/user2/openpilot/laika_repo'...
Cloning into '/home/user2/openpilot/opendbc'...
Cloning into '/home/user2/openpilot/panda'...
Cloning into '/home/user2/openpilot/rednose_repo'...
Submodule path 'cereal': checked out '626679d1a56c689cbad7bf99a08c5ecbf9d073fb'
Submodule path 'laika_repo': checked out '765c6584c3d7f27e1af0f1180cc29766d5319f09'
Submodule path 'opendbc': checked out 'b7cf1a67bc71b674e6793ba1f2fff5d29fee1e6b'
Submodule path 'panda': checked out '8b41ed3b81b0315a5f9c9aa0cad84bfaef91f3f4'
Submodule path 'rednose_repo': checked out '2e556b8219185708ed974a4b6502796607d7ce0d'
ubuntu_setup.sh: line 81: pyenv: command not found
ubuntu_setup.sh: line 82: pyenv: command not found
ubuntu_setup.sh: line 83: pyenv: command not found
ubuntu_setup.sh: line 88: pip: command not found
ubuntu_setup.sh: line 91: pipenv: command not found

```

**Expected behavior**

Docker build script succeeds.

Operating system: [e.g. Ubuntu 16.04]
Ubuntu 20.04

LoganxDev, 2020-09-28:
What happens if you run it one directory up instead of the tools directory?

AnEthernetCable, 2020-09-30:
This can be fixed by adding the following to `~/.profile`:
```
export PATH="$HOME/.pyenv/bin:$PATH"    
eval "$(pyenv init -)"
eval "$(pyenv virtualenv-init -)"
```

Perhaps the setup script should add this automatically? It appears the pyenv installer the script uses does not do this.

LoganxDev, 2020-09-30:
Hm, it looks like it is doing that as part of `openpilot_env.sh`, different shell being used may be causing the issue?

AnEthernetCable, 2020-10-01:
No matter if it's in both `~/.profile` and `~/.bashrc`, OR just `~/.profile` it should work. It could also add these variables to the current terminal session so a restart of the session is not required.

AnEthernetCable, 2020-10-02:
If we add `source ~/.bashrc` between line 80 and 82 of `ubuntu_setup.sh`, this issue should be fixed.

adeebshihadeh, 2021-02-02:
You need to make sure pyenv is in your path.

closed 2021-02-02 by adeebshihadeh