commaai/openpilot #2139: New driver monitoring model uncertain at night with EON

ShaneSmiskol, 2020-09-07:
**Describe the bug**

At night with the EON, I get `CHECK DRIVER FACE VISIBILITY` alerts about every 15 to 30 seconds. It seems most common when a light shines inside the car like from headlights or when going through intersections with traffic lights.

**How to reproduce or log data**

Use an EON at night on 0.7.8

**Expected behavior**

Before the driver monitor model update in 078, the model recognized that it couldn't see the driver properly at night and instead switched to the ~1 min wheel timer (or however long it is). So expected behavior is to disable DM at night when it can't see driver, and don't display `CHECK DRIVER FACE VISIBILITY` alerts every few seconds.

**Device/Version information (please complete the following information):**
 - Device: EON Gold
 - Dongle ID: e010b634f3d65cdb
 - Route: Don't have any with my driver view recorded. I can turn on and get a ddrive on request
 - Timestamp: NA
 - Version: 0.7.8
 - Car make/model: 2017 Toyota Corolla

xuimod, 2020-09-08:
Thank you.   I had the exact same problem.  

Version 0.7.8 with a 2017 Toyota Prius.

Doesn't happen during the day.  Only happened at night with the update to 0.7.8.  Never saw the 'new driver monitoring model uncertain' ever before this.  Never happened and never saw that message in previous versions.  And it is extremely annoying.... Openpilot actually forciably disengaged and displayed a message in a red background saying something like, system disengaged, take control immediately.  That red message probably occured 7-8 times on my drive home from work this morning (it was dark, i drive home at about 5am) and the 'new driver monitoring model uncertain' occurred i would guess 50-100 times during my commute back home.

my commute back home is about 28 miles (20 of that is on the highway) and takes about 30 minutes.

Also, not sure if this is relevant but I notice the small yellow triangle that appears on the Eon when the car ahead of you is detected is no longer present.  I found the small yellow triangle to be very useful.  But with version 0.7.8 the small yellow triangle no longer appears either at night or during the day.

ZwX1616, 2020-09-08:
This is expected when ANY lighting is present. The new model is actually better at detecting faces in the dark than before, and the uncertain message displays when face is detected but visibility is poor. Real solution will be using IR at night. (hint: comma two;))

pd0wm, 2020-09-08:
@xuimod The driver monitoring alert should be silent. If you got a red/loud alert that was caused by a different issue. Can you share the drive where that happened?

Regarding the triangle. If the stock system is controlling longitudinal, we are now hiding the triangle. It could be misleading if it was showing the triangle, but the stock system didn't detect the lead car yet since the openpilot detection is completely separate from the stock one.

xuimod, 2020-09-08:
> This is expected when ANY lighting is present. The new model is actually better at detecting faces in the dark than before, and the uncertain message displays when face is detected but visibility is poor. Real solution will be using IR at night. (hint: comma two;))

> This is expected when ANY lighting is present. The new model is actually better at detecting faces in the dark than before, and the uncertain message displays when face is detected but visibility is poor. Real solution will be using IR at night. (hint: comma two;))

The system was working great before.  Pushing updates that gimp older hardware almost forcing users to upgrade doesn't sit well with me.  **I bought my Eon less than a year ago**.  

And will the same happen to Comma two if/when a Comma three is released?

Is there anyway to revert to an older version like 0.7.7 and turn off auto updates?

ZwX1616, 2020-09-08:
> > This is expected when ANY lighting is present. The new model is actually better at detecting faces in the dark than before, and the uncertain message displays when face is detected but visibility is poor. Real solution will be using IR at night. (hint: comma two;))
> 
> > This is expected when ANY lighting is present. The new model is actually better at detecting faces in the dark than before, and the uncertain message displays when face is detected but visibility is poor. Real solution will be using IR at night. (hint: comma two;))
> 
> The system was working great before. Pushing updates that gimp older hardware almost forcing users to spend over $1000 to upgrade doesn't sit well with me. **I bought my Eon less than a year ago**.
> 
> And will the same happen to Comma two if/when a Comma three is released?
> 
> Is there anyway to revert to an older version like 0.7.7 and turn off auto updates?

The intention of the alert is to notify the user if the device is mounted poorly. We will make it less annoying for EON users at night.

ShaneSmiskol, 2020-09-08:
It might still be a bug, as I fully covered my face with my hands but the uncertain alert still appeared a few times, even though it definitely couldn't see me due to my hands. Previously if I covered my face it would just not see a face. But the alert showing up at night when I cover my face makes it seem like there's something else going wrong

ZwX1616, 2020-09-08:
> It might still be a bug, as I fully covered my face with my hands but the uncertain alert still appeared a few times, even though it definitely couldn't see me due to my hands. Previously if I [...]

It doesn't always take a face to detect a face. Humans can look at somebody wearing a helmet and think "hmm, there is probably a face (because I see their neck/chest), but I am really not sure what's going on under it". In this case the model does exactly that.

xuimod, 2020-09-09:
> The system was working great before. Pushing updates that gimp older hardware almost forcing users to spend over $1000 to upgrade doesn't sit well with me. **I bought my Eon less than a year ago**. [...]

I ended up installing a Open Pilot fork on my Eon.

For future reference, if anybody decides to go this route a utility called Workbench makes it very easy to install an Open Pilot fork on your Eon.

Installing a Fork of Openpilot with Workbench:  https://medium.com/@jfrux/installing-a-fork-of-openpilot-with-workbench-de35e9388021

ShaneSmiskol, 2020-09-09:
I'll see if I can get out of the camera view completely and see what the model does, if it still gives the uncertain warning or not.

closed 2020-09-14 by adeebshihadeh