commaai/openpilot #2142: Hyundai SCC automatic resume is unreliable

gregjhogan, 2020-09-08:
**Describe the bug**

Somewhere in the range of 1% to 10% of the time the car will not accept the resume command from openpilot for 5+ seconds

**How to reproduce or log data**

Sit behind a parked car, engage, and let openpilot send the resume command repeatedly when the car goes into the state where it says press pedal or accel button to resume.

**Expected behavior**

The car always accepts the resume command from openpilot within a second and goes back into the state where it is prepared to accelerate immediately when the lead vehicle pulls away.

**Device/Version information (please complete the following information):**
 - Route: You should be able to easily reproduce this issue, if not I can record and provide a route
 - Timestamp: n/a
 - Version: 01111b3f28c748769db8f27646d9086f00a417ef
 - Car make/model: 2020 Hyundai Palisade

**Additional context**

https://github.com/commaai/openpilot/blob/e8c8e9dab25c81fbeefa4fa05bf38166f9abe203/selfdrive/car/hyundai/carcontroller.py#L79-L80

Most of the time the car will accept the very first resume command openpilot sends, but some percentage of the time it takes way too long and you have to manually resume.

What I have tried:
- making the parity bit always correct, and always incorrect and neither seemed to change the reliability (so it seems like the parity bit doesn't matter)
- making the counter always the last counter + 1, sending 15 messages all at once incrementing the counter for each message (so it wraps around back to the next value the car would send) and always sending the counter - 1 which should always be incorrect and nothing seemed to change the reliability (so it seems like the counter doesn't matter)
- increasing the frequency all the way up to 100 Hz, which causes the car to fault and engage the parking break sometimes, and it seems like the higher you increase the frequency the less reliable it becomes

The vehicle sends the button state at 50 Hz so perhaps sometimes the next message from the vehicle will come in very close after openpilot sent a message, overriding what openpilot sent?

I would guess this is an issue for the Sonata and Palisade since they use the same radar and this is related to SCC, and perhaps other vehicles, too.

xps-genesis, 2020-09-10:
@gregjhogan give the changes in #2152 a try. seems to be very reliable have a sonata 2020 user who seems to be happy.
you can remove the lead relative velocity check in the PR if you want to just test activating behind a target in a parking lot to see if the resume spam works every time the cluster shows display to it resume.

gregjhogan, 2020-09-12:
Someone reproduced the issue and verified that before adding this code there is a problem and after this change it is fixed?

xps-genesis, 2020-09-12:
I have been working with people on button spamming to auto change set speed. we tried to use the 5Hz spam like it is in Master, and it was unreliable or it will fault out. so I played with the logic and this logic in PR works every time no issues.

so in general the new logic in the PR works for button spam. it works flawlessly for changing set speed from 20mph to 80mph or down from 85 to 20mph through spamming without any faults. Since it worked for change in set speed, and you brought the issue up about unreliable auto resume, i proposed the same logic for auto resume that I used.

i have added lead vRel check so the resume spam doesn't happen when the lead is creeping forward for no reason. this ensure resume spam only happens when lead accelerates.

xps-genesis, 2020-09-18:
@gregjhogan has this been tested?

xx979xx, 2020-09-18:
https://github.com/commaai/openpilot/commit/b81f404457b2bfeee1d9c18c86af2aeb167b9403 is bad, revert fix it for me

gregjhogan, 2020-09-19:
@xps-genesis @xx979xx I tested both and I still have instances where it takes more than 10 seconds for a resume message to get accepted

xx979xx, 2020-09-19:
@gregjhogan the issue, as I understand it, cluster instrument sends `clu11` at 50 Hz, `SCC` has msg count threshold to accept that button pressed and has  time window for accepting Rx msg which shifts as soon as valid msg received .  it is a timing issue, and defining this time window solve it.
the community way is the most effective up to now

gregjhogan, 2020-09-29:
I have tested everything suggested here and it would be inaccurate to say anything suggested fixes this issue

closed 2020-10-15 by adeebshihadeh