commaai/openpilot #2210: Poor Performance with Black Lane Markers

C0mpl3t3N00b, 2020-09-21:
**Describe the bug**
Lines on the highway are painted black (seem to be painted over) but lane reflectors are still present, and seem to have enough contrast that I and other drivers can tell where lane markers are . When Comma is activated it does not see lane markers and will sometimes ride over the lines or get very close to the lane marker it does not see.

**How to reproduce or log data**
Driving South Bound on Hwy 101 from Morgan Hill, CA to Gilroy, CA will reproduce this consistently. 

**Expected behavior**
If I and all other drivers can see the lane dividers then Comma should also "see" the lane markers too, so that my vehicle can stay centered in the lane and not begin to ride over the line.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID :bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e|2020-09-20--12-01-31
 - Timestamp: Around 35mins into the drive.
 - Version: v0.7.9 master-ci 9fe3d286a641a23431f4487c8a5980be4d8de4bc
 - Car make/model: 2020 Toyota Rav4 Hybrid LE

**Additional context**

[image]

grekiki, 2020-09-23:
If everything goes as planned should be fixed in 0.8

closed 2020-09-23 by grekiki

C0mpl3t3N00b, 2020-11-15:
@grekiki can we reopen this issue as I have been experiencing it still with v0.8 with segment below:

bdad50b8a1f9fa8e|2020-11-14--12-08-53--25
[image]