commaai/openpilot #2215: Merge sensord and qualcomm sensor daemon

adeebshihadeh, 2020-09-21:
Currently, sensord and sensors.qcom use about 10% of a core. sensord should be able to work without the qualcomm  daemon and without going through android.

adeebshihadeh, 2020-09-21:
@deanlee are you interested in looking into this one?

deanlee, 2020-09-22:
I'll try.is there any information or tips I need to know?

adeebshihadeh, 2020-09-22:
This is only for sensors_qcom.cc. Here's an overview of the android sensor stack - https://source.android.com/devices/sensors/sensor-stack. I believe all the sensor drivers should be here https://github.com/commaai/android_kernel_comma_msm8996/tree/master/drivers/input/misc. Should be a fun project :)

ihsakashi, 2020-09-22:
https://source.codeaurora.org/quic/la/platform/hardware/qcom/sensors/. This is Qualcomm's sensor HAL implementation. I do not know if these go all the way back to marshmallow but they should help

deanlee, 2020-09-23:
@ihsakashi  thank you!
I think we're going to bypass the entire android sensor framework .but this link is still hepful to me.

ihsakashi, 2020-09-23:
Np, bypassing is going to be better

deanlee, 2020-10-12:
@adeebshihadeh Is this sensor list correct?
LSM6DS3 
AK09911
LTR579
Are you sure all the drivers are in here?: https://github.com/commaai/android_kernel_comma_msm8996/tree/master/drivers/input/misc.

closed 2021-01-16 by adeebshihadeh