commaai/openpilot #2227: Honda can't get right result

LOVEChen, 2020-09-25:
**Describe the bug**
can't get right fw version result

**How to reproduce or log data**
term 1 log:
cd /data/openpilot/selfdrive/boardd
pkill ./manager.py OR killall tmux
./boardd
selfdrive/boardd/boardd.cc: starting boardd
selfdrive/boardd/boardd.cc: attempting to connect
selfdrive/boardd/boardd.cc: fw signature: 0a8e**********
selfdrive/boardd/boardd.cc: panda serial: 7569**********
selfdrive/boardd/boardd.cc: connected to board
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/boardd.cc: Receive buffer full
selfdrive/boardd/pigeon.cc: panda GPS start
selfdrive/boardd/pigeon.cc: panda GPS on
sendcan: Reader was evicted, reconnecting
selfdrive/boardd/boardd.cc: missed cycles (0) -773110
selfdrive/boardd/boardd.cc: missed cycles (0) -2149842
selfdrive/boardd/boardd.cc: missed cycles (0) -177343
selfdrive/boardd/boardd.cc: missed cycles (0) -35659
selfdrive/boardd/boardd.cc: missed cycles (0) -733221
selfdrive/boardd/boardd.cc: missed cycles (0) -274734
selfdrive/boardd/boardd.cc: missed cycles (0) -147030
selfdrive/boardd/boardd.cc: missed cycles (0) -69152
selfdrive/boardd/boardd.cc: missed cycles (0) -757290
selfdrive/boardd/boardd.cc: missed cycles (0) -167025
selfdrive/boardd/boardd.cc: missed cycles (0) -7238
selfdrive/boardd/boardd.cc: missed cycles (0) -2005781
selfdrive/boardd/boardd.cc: missed cycles (0) -247024
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/boardd.cc: missed cycles (0) -3956094
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
term 2 log:
python fw_versions.py --scan
Getting vin...
vin query retry (1) ...
vin query retry (2) ...
vin query retry (3) ...
vin query retry (4) ...
vin query retry (5) ...
vin query retry (6) ...
vin query retry (7) ...
vin query retry (8) ...
vin query retry (9) ...
vin query retry (10) ...
VIN: 00000000000000000
Getting VIN took 1.080 s

fw
{
}

comma power is connected 
hardware: black panda officail version
panda firmware version : v1.7.5-EON-DEBUG AND v1.7.3-EON-DEBUG
i aslo used this 2 version panda firmware 
still can not get right result

term 2 always output vin: 000000000
and term 2 always Receive buffer full AND Transmit buffer full

Operating system: [MACOSX 10.15.6 with SecureCRT  OR workbench ]

adeebshihadeh, 2020-10-22:
Are you sure the panda is in a valid safety mode for FW query? You can also get your VIN and FW versions through useradmin if select a route and open the first segment log.

LOVEChen, 2020-10-22:
comma.ai resolve my problems

closed 2020-10-22 by LOVEChen