commaai/openpilot #2228: Braking

kyeli2, 2020-09-25:
I have noticed when coming up to stopped cars my open pilot doesn't see the stopped cars.  But if I'm following cars it stays with the cars in front, stop or go, great.  Why doesn't it recognize cars that are stopped?

Kyle

Consider these options before opening an issue for a question:

- checking the FAQ at https://comma.ai/faq
- checking the wiki at https://wiki.comma.ai
- asking your question on our community discord at https://discord.comma.ai

-->

VirtuallyChris, 2020-09-25:
Hi Kyle,

What car do you have?

adeebshihadeh, 2020-09-28:
Closing as this should be a bug report, not a question. Feel free to reopen using the bug report template. Also make sure to check the README and ensure openpilot handles longitudinal control on your car.

closed 2020-09-28 by adeebshihadeh