commaai/openpilot #2233: Adaptive cruise control disengages when braking due to car in front braking on 2018 Nissan Leaf, with seatbelt warning shown on HUD

willemcvu, 2020-09-27:
To replicate:

1. Have openpilot activated for steering, with Nissan stock system doing longitudinal control.
2. Have lead car indicated on HUD with set speed faster than lead car speed.
3. Lead car brakes for a complete stop.

At this point, the Nissan system starts braking, but at about 20mph, 9/10 times adaptive cruise control disengages, openpilot disengages, and the HUD shows a "seatbelt warning" saying that the system can't be engaged if the seat belt is not connected (I can't remember the exact message text).

It is quite jarring for all adaptive cruise control to suddenly disengage mid-braking :/ 

With the stock Nissan adaptive cruise control, this (obviously) does not happen.

pd0wm, 2020-09-28:
Can you fill out the bug report template with dongle id + route?

willemcvu, 2020-09-28:
Yep, I'll do that when I get home tonight - I see the template now, apologies for not following the format the first time around.

willemcvu, 2021-01-01:
See new bug report at #19631. Closing.

closed 2021-01-01 by willemcvu