commaai/openpilot #2251: [Enhancement] Yellow "Hold Steering Wheel" warning on Toyota MFD / HUD

cydia2020, 2020-10-01:
Hi,

When the driver lets go of the steering wheel while using Toyota's factory LKA system, the system initially displays a white "Hold Steering Wheel" warning on the MFD / Dash; if the driver failed to respond, the system will proceed to display another yellow coloured "Hold Steering Wheel" warning on both the MFD / Dash and the HUD to alert the driver.

I am think that this can be used by openpilot to warn users about scenarios that would require an immediate input.

For example:
1. When the turn exceeds the steering limit but the driver failed to respond for a set amount of time
2. When the driver monitoring distraction level reaches promptDriverDistracted
3. When the system disengages unexpectedly

Unfortunately I am unable to find an image that matches my description but this warning does exist. I can try to take a photo of it some day if I think it is safe to do so.

cydia2020, 2020-10-01:
Also while we are on the topic of alerts, I think VisualAlert.steerRequired on some events are not necessary and could potentially be a distraction. Notably on EventName.laneChange, and EventName.laneChangeBlocked

cydia2020, 2020-10-02:
By the way, from a human factor point of view, may I ask the reason behind _preDriverDistracted_ is **Driver Distracted** while _promptDriverDistracted_ is **Driver Appears Distracted** ?

Shouldn't the warning text of these two be flipped?

pd0wm, 2020-10-06:
I want to keep the HUD warnings generic between all car brands, so don't want to introduce a separate level that's only supported on certain cars. We could consider switching the alert from the white to the yellow one.

Please post the other two complaints in this issue: https://github.com/commaai/openpilot/issues/2271. We're doing a big cleanup of all the alerts, so we can look into those as well.

cydia2020, 2020-10-06:
> I want to keep the HUD warnings generic between all car brands, so don't want to introduce a separate level that's only supported on certain cars. We could consider switching the alert from the [...]

Ok I'll put the other alert related problems to the other issue, thanks

adeebshihadeh, 2020-10-19:
I think if we're going to stick to one or the other, we'll probably want to stay with the yellow one considering what we use it for.

closed 2020-10-19 by adeebshihadeh