commaai/openpilot #2271: Alert cleanup project

pd0wm, 2020-10-05:
* Alerts that need to be permanent?
* Alerts that are too annoying, and need improvements on trigger?
* Other alert improvements?

xps-genesis, 2020-10-06:
Issue:
if OP is diengaged in the middle of an alert display, the alert is repeated after disengagement.
Example:
Vehicle on a curve that exceeds the actuator limits. Alert displayed "Take Control, steering limits exceeded". At this time if the OP is disabled by pressing brake pedal, there is a OP disengage chime(soft disable) followed by repeatation of a quick "Take Control, steering limits exceeded" with chime.

cydia2020, 2020-10-06:
Hi, as instructed, I am putting my take on the alerts here

I think VisualAlert.steerRequired on some events are not necessary and could potentially be a distraction. Notably on EventName.laneChange (no need to display hold steering wheel alert after the user input has already been given), and EventName.laneChangeBlocked

And on the topic of alerts, may I ask the reason of having preDriverDistracted being Driver Distracted while promptDriverDistracted is Driver Appears Distracted ?

Shouldn't the warning text of these two be flipped?

adeebshihadeh, 2020-10-06:
@xps-genesis do you have a route where this happened?

adeebshihadeh, 2020-10-06:
TODO:
* [ ] fix steer saturation alert
* [ ] reevaluate DM alerts
* [ ] clear irrelevant alerts on state transition

VirtuallyChris, 2020-10-07:
Add an alert if most fw versions match, but one or two are different, don’t show generic ‘Car unsupported’ alert. Show something like ‘Car likely supported’ ‘Verify fw versions’ - this is a pretty frequent cause of returns since people think their car isn’t supported (edited)

xps-genesis, 2020-10-07:
> @xps-genesis do you have a route where this happened? Did you disengage immediately when seeing the alert? This is probably because the minimum duration of the steer saturated alert is 3s.

@adeebshihadeh that is exactly why it is happening. but isn't that a nuisance that the steer saturated error reappears after disengagement of OP? not just this alert,  after a disengagement event, any timer on the previous alert should be reset so it doesn't reappear. worst part is the chime repeats too. sometime the disengagement chime and the repeat alert chime partially overlap causing a distortion in sound.

I don't have the route from Friday when I last saw, I can redo the same road and repeat it if the route is really required.

adeebshihadeh, 2020-10-11:
@xps-genesis No route needed, fixing this in #2318. The sound distortion is different issue that I'll look into fixing for this release too.

closed 2020-10-19 by adeebshihadeh