commaai/openpilot #2279: Build python on device with optimizations -- $100+ bounty

adeebshihadeh, 2020-10-06:
$100 for a merged PR against [commaai/eon-neos-builder](https://github.com/commaai/eon-neos-builder) that builds python 3.8.2 on a NEOS device with optimizations enabled (e.g. `--enable-optimizations`). Additional $10 for every percent of a core we get back, as measured by the average sum from `selfdrive/debug/cpu_usage_stat.py`.

Starting point - https://github.com/commaai/termux-packages/tree/neos

pd0wm, 2020-10-06:
https://github.com/commaai/eon-neos-builder/blob/master/devices/eon/home/install.sh is the script that we run on the device after generating the image that handles all the on device build steps. After that we take an image from the device.

nanamiwang, 2020-10-08:
Any plan running OP under PyPy or Numba? I had a try porting OP to PyPy on EON, compilled PyPy on EON successfully, failed to run controlsd because PyPy doesn't support numpy officially. I will continue the work if this make sense. BTW, is it possible that we remove OP dependency on numpy? seems we don't have too much code depending on numpy.

closed 2021-03-09 by adeebshihadeh