commaai/openpilot #2289: Comma.ai won't turn on!

peterk714, 2020-10-07:
*My Comma.ai won't turn on!
I have remove my Comma.ai and connected to power cord.  I have tried repeatedly to turn on the device and there is no response.
Please advise?

cydia2020, 2020-10-07:
from #2290 it appears to be a boot loop, contact support?

LoganxDev, 2020-10-07:
From the Wiki:

> When using a wallcharger at your desk, make sure it can output 2-3A (installation draws just a little above 1.0A).

Does your power source meet that requirement?

peterk714, 2020-10-07:
Yes it meets that voltage requirement.  I used 60 watt usb to C laptop charger.  I used that same charger to install the software on the AON when I first got it 12 months ago.

Peter

peterk714, 2020-10-07:
I’m going on a long trip this Friday. I would like to have my problem resolved by then.

Thanks,

Peter

closed 2020-10-07 by vanillagorillaa

peterk714, 2020-10-07:
45 watt 2.4 amp output.  I used this same charger to configure the Eon when it first arrived

peterk714, 2020-10-07:
When is somebody from Comma.ai going to provide me a resolution to my problem????

vanillagorillaa, 2020-10-07:
Hello Peter, a response email was sent to you earlier this morning.

peterk714, 2020-10-07:
It said contact support

vanillagorillaa, 2020-10-07:
Yes, your support email was responded to earlier this morning by the support team.

peterk714, 2020-10-07:
I have two c connector power sources, 60 Watt Lenovo Laptop and universal 45 watt source and they are both over 2.4 amps

peterk714, 2020-10-07:
How do I turn this thing on????